betaflight/betaflight #7980: Caddx Turtle v2 uart problem (Cannot start recording via aux channel)

milikiller, 2019-04-12:
**Describe the bug**
I have Caddx turtle V2 and I tryed to use uart for camera control.  I use Betaflight 4.0  on BetaFPV F4 FC (target OmnibusF4SD)

Navigation in OSD Working very well through sticks. But start / stop video recording via aux chanel not working for me.

I tryed CAMERA POWER in Modes set to aux 1 same way as I use ARM because I want to start video recording when I ARM the copter.  But nothing happend. No blinking red light on top corner...

**Flight controller configuration**
```# Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Apr 11 2019 / 07:31:36 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id 

# name
name Beta 85X

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature LED_STRIP

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 16384 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# led
led 0 0,0::A:0
led 1 1,0::A:0
led 2 2,0::L:0
led 3 3,0::L:0

# color

# mode_color
mode_color 6 0 2

# aux
aux 0 0 0 1200 2100 0 0
aux 1 1 2 900 1200 0 0
aux 2 13 5 1250 1750 0 0
aux 3 28 0 1200 2100 0 0
aux 4 33 0 1500 2100 0 0
aux 5 35 1 1700 2100 0 0

# adjrange
adjrange 0 0 2 1350 1650 12 2 0 0

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_lowpass_type = PT1
set acc_calibration = 58,44,-87
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set fpv_mix_degrees = 35
set serialrx_provider = IBUS
set blackbox_device = NONE
set dshot_idle_value = 1300
set motor_pwm_protocol = DSHOT1200
set vbat_min_cell_voltage = 350
set vbat_warning_cell_voltage = 360
set current_meter = NONE
set vbat_scale = 108
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 3
set yaw_deadband = 5
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_vbat_pos = 385
set osd_tim_2_pos = 2387
set osd_flymode_pos = 2124
set osd_throttle_pos = 2405
set osd_ah_pos = 2126
set osd_mah_drawn_pos = 344
set osd_craft_name_pos = 394
set osd_warnings_pos = 2345
set osd_avg_cell_voltage_pos = 2373
set vtx_band = 3
set vtx_channel = 8
set vtx_power = 2
set vtx_freq = 5945

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 200
set anti_gravity_gain = 2000
set feedforward_transition = 30
set iterm_relax_type = GYRO
set iterm_limit = 500
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set p_pitch = 60
set i_pitch = 80
set d_pitch = 50
set f_pitch = 80
set p_roll = 60
set i_roll = 80
set d_roll = 50
set f_roll = 80
set p_yaw = 60
set i_yaw = 75
set f_yaw = 85
set angle_level_strength = 75
set horizon_level_strength = 75
set level_limit = 60
set abs_control_gain = 10
set d_min_roll = 38
set d_min_pitch = 38
set d_min_boost_gain = 30
set d_min_advance = 80

# rateprofile
rateprofile 0

set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 177
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0

# end the command batch
batch end```

Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Apr  1 2019 / 11:35:37 (88665a678) MSP API: 1.41
Caddx Turtle V2 FW march_2019 from BF git site

milikiller, 2019-04-12:
Sorry, my mystake. I reflashed camera again with latest firmware and everything is working

GhostMaster9, 2019-04-12:
Close the issue them...

closed 2019-04-12 by McGiverGim