betaflight/betaflight #7981: Betaflight 4.0 Jeti EXBus not working on FURYF4OSD

xoration, 2019-04-12:
I just flashed Betaflight 4.0.0 on my FURYF4OSD and EXBus is not working anymore (No channels moving no response).
When I flash 3.5.7 EXBus is working fine.

# version
# Betaflight / FURYF4OSD (FY4O) 4.0.0 Apr 11 2019 / 07:16:52 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map
map TAER1234

# serial
serial 2 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_calibration = -80,-20,-9
set serialrx_provider = JETIEXBUS
set small_angle = 180

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end

RoadyFPV, 2019-04-12:
Hi i use Jeti ExBus to, have no problems, Board is OMNINXTF7 and BrainFPV RADIX

etracer65, 2019-04-12:
@xoration What UART do you have the receiver connected to? Also, please provide the CLI output for the `serial` command. It doesn't look like you've enabled the Serial Rx switch on any of them.

xoration, 2019-04-12:
@etracer65 It is on UART1 output of serial is:
# serial
serial 20 1 115200 57600 0 115200
serial 0 64 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

In the betaflight configurator Serial RX is enabled for UART1

etracer65, 2019-04-12:
There weren't any changes that should have affected Jeti EXBus. Can you also please provide a `diff` of your settings that work on 3.5.7?

xoration, 2019-04-12:
```
# diff

# version
# Betaflight / FURYF4OSD (FY4O) 3.5.7 Mar 15 2019 / 21:56:07 (e9130527c) MSP API: 1.40

board_name FURYF4OSD
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map
map TAER1234

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set serialrx_provider = JETIEXBUS

# profile
profile 0

# rateprofile
rateprofile 0

```

xoration, 2019-04-12:
I just reflashed Betalfigh 4.0 just changing the serialrx_provider an channel mapping but same result it does NOT work.

```
# version
# Betaflight / FURYF4OSD (FY4O) 4.0.0 Apr 11 2019 / 07:16:52 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map
map TAER1234

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set serialrx_provider = JETIEXBUS

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end

```

etracer65, 2019-04-13:
A couple more things to check between your working 3.5.7 version and the 4.0 version. What are the values for these from each version?
```
get serialrx_inverted
get serialrx_halfduplex
```
That flight controller appears to have a programmable inverter on UART1 so if that was enabled then Jeti EXBus wouldn't work. You could also confirm by connecting your receiver to a different UART.

xoration, 2019-04-13:
On 3.5.7
# get serialrx_inverted
serialrx_inverted = OFF

# get serialrx_halfduplex
serialrx_halfduplex = OFF

On 4.0 it's the same both are off. I can't connect the receiver to a different uart as this is one of those tiny micro builds and I literally would have to disassemble everything to resolder the receiver. I rather stay on 3.5.7 forever. :)

mikeller, 2019-04-14:
@xoration: There was a change in how the JetiEXBus protocol works: #6767 Maybe @marv-t can help here - this might even be worthy of being added to the wiki.

RoadyFPV, 2019-04-14:
Did u mark the serial rx box  on uart 1 in the configurator?

marv-t, 2019-04-14:
@xoration 
At which pin the receiver is exactly connected.
Is telemetry used?

xoration, 2019-04-15:
I am not using telemetry yet I am just trying to get the channels to work so to anwear your question, no.
At which pin well from the looks of it it's the PPM/SBus pin whatever this is. The quad I am talking about is a Diatone-2018-GT-R90 in it's default configuration.

marv-t, 2019-04-21:
@xoration 
Before 4.0, the receiver was connected to RX. If the telemetry was to be used, a resistor had to be installed between RX & TX.
Now it is at 4.0 so that only the TX port is used. If a resistor is installed, the receiver works as usual with the old setup.
Unfortunately I have not considered the case where only the RX port is used.
 
@Mikeller
Should I write a routine that checks where the receiver is connected and then properly configure the Uart.

mikeller, 2019-04-29:
@marv-t: Yes please.

Skym4x, 2019-10-02:
hi @xoration have find a solution for Jeti ?
thank

simored5, 2020-01-09:
Hello. I am having same issue on a hobbywing fc that uses furyf4osd.

I started with the latest version of BF ad could not get my receiver to connect and have tried numerous BF 4+ versions but can only get 35.7 or Lower to work.

Is there an expected update to resolve this issue.

Thanks for your help

mikeller, 2020-01-12:
@simored5: See https://github.com/betaflight/betaflight/issues/7981#issuecomment-485233008 - if your setup works with 3.5.7 then it is likely that your RX is connected to the RX pin, and you need to move it to the TX pin of the UART to make it work in 4.0 and up.

simored5, 2020-01-13:
Thanks for your reply. I had already managed to figure that out and everything is working now.

closed 2022-05-10 by haslinghuis