betaflight/betaflight #7982: Add rate_limit settings to MSP API version 1.42

ianrmurphy, 2019-04-12:
Allows for new cli settings `rate_limit_roll`, `pitch_rate_limit` & `yaw_rate_limit` to be supplied to the configurator for visualisation.

ianrmurphy, 2019-04-12:
I'm willing to develop this feature, but since it needs co-ordination with any other MSP API v1.42 changes, I need to understand the collaboration process.

mikeller, 2019-04-14:
@ianrmurphy: The collaboration process is simple really: Any pull request slated for Betaflight 4.1 that needs changes in configurator as well will be labelled as such, and needs to reference the pull request for configurator that contains the required change. We will then take care of merging these atomically.

ianrmurphy, 2019-05-14:
Closed as merged.

closed 2019-05-14 by ianrmurphy