betaflight/betaflight #7983: Eval bf.lua update requirement note for 4.0.x. - (RTC ticks from OpenTX bf.lua)

pitts-mo, 2019-04-12:
Please evaluate addition of bf.lua update requirement notes for 4.0.x.

My specific example reason:
After update to BF 4.0.0 release DateTime was not initializing correctly in my OSD. (Taranis x9d+, OpenTX 2.2.3 release, bf.lua 1.1.0)

I began to suspect the cause was RTC change for ticks support and found related change already committed :-) ( thanks @mikeller - https://github.com/betaflight/betaflight-tx-lua-scripts/pull/159) 

A new download/build of bf lua from source less than an hour ago did restore my OSD DateTime to initialize correctly. :-)  (https://github.com/betaflight/betaflight-tx-lua-scripts)

pitts-mo, 2019-04-13:
if you are using the Betaflight TX Lua Scripts, note that updates to Betaflight TX Lua Scripts since 1.1.0 add new functionality and compatibility (Real Time Clock - ticks) for Betaflight 4.0. Please update to the next release as it becomes available ( installation and pre-release building instructions [here](https://github.com/betaflight/betaflight-tx-lua-scripts) )

mikeller, 2019-04-14:
@pitts-mo: Good point. A release of the betaflight-tx-lua-scripts to match Betaflight 4.0 is still outstanding, we'll try to get this done within the next few days, and add an update notice to the release notes.

laz2wiringpi, 2019-04-15:
FYI . nano diversity receiver Make sure you are using CROSSFIRE  2.94 BETA for LUA to work

i have been driving myself nuts  with both the  crsf and  lua scrips not working.

pitts-mo, 2019-04-30:
Thank you @mikeller . I see you perform a great service across the betaflight related communities. With great respect and appreciation to you and other developers working in betaflight related projects...

Thank you all so very much for keeping our family moving forward on our path of learning and improvement.

closed 2019-04-30 by pitts-mo