betaflight/betaflight #7987: When setting motor_pwm_protocol to dshot600, Gyro light goes off and Sonar light turns on

mmakam2, 2019-04-12:
I have a CLRACINGF4 (bardwell), I noticed today when i changed the motor pwm protocol from default to dshot600, the gyro light went off and the sonar light came on. Really weird, not sure if its a configurator issue or firmware issue. It flies fine otherwise.

'''
# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Apr  3 2019 / 13:44:35 (22b9f3453) MSP API: 1.41

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id 

# name
name RAGINGDRONER

# resources
resource BEEPER 1 NONE
resource MOTOR 1 B01
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 A03

# timer
timer B00 1
timer B01 1
timer B04 NONE
timer B08 NONE

# dma
dma pin B04 NONE
dma pin B08 NONE

# mixer
mixer CUSTOM
mmix 0  1.000 -0.850  0.880 -1.000
mmix 1  1.000 -1.000 -0.910  1.000
mmix 2  1.000  0.850  0.910  1.000
mmix 3  1.000  1.000 -0.880 -1.000

# servo

# servo mix

# feature
feature ESC_SENSOR

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1925 2100 0 0
aux 1 35 1 1900 2100 0 0
aux 2 49 1 1400 1575 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 7 s 1000

# display_name

# master
set gyro_lowpass_type = PT1
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_lpf_gyro_min_hz = 140
set dyn_lpf_gyro_max_hz = 600
set acc_calibration = 108,19,97
set rssi_channel = 8
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set dshot_idle_value = 500
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set vbat_max_cell_voltage = 435
set current_meter = ESC
set battery_meter = ESC
set vbat_scale = 111
set beeper_frequency = 0
set small_angle = 180
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_pos = 2337
set osd_tim_2_pos = 2071
set osd_vtx_channel_pos = 2528
set osd_craft_name_pos = 2057
set osd_avg_cell_voltage_pos = 2304
set debug_mode = GYRO_SCALED
set cpu_overclock = 216MHZ
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5658

# profile
profile 0

set dyn_lpf_dterm_min_hz = 105
set dyn_lpf_dterm_max_hz = 200
set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 105
set feedforward_transition = 10
set smart_feedforward = ON
set iterm_relax = RPY
set iterm_relax_type = GYRO
set p_pitch = 35
set d_pitch = 24
set d_roll = 37
set f_roll = 75
set p_yaw = 30
set i_yaw = 0
set d_yaw = 20
set f_yaw = 60
set abs_control_gain = 10
set use_integrated_yaw = ON
set d_min_pitch = 18
set d_min_boost_gain = 35
set launch_control_mode = PITCHONLY
set launch_trigger_allow_reset = OFF
set launch_angle_limit = 13

# rateprofile
rateprofile 0

set roll_rc_rate = 110
set pitch_rc_rate = 110
set yaw_rc_rate = 118
set roll_srate = 76
set pitch_srate = 76
set yaw_srate = 72
set tpa_rate = 70
set throttle_limit_type = SCALE

# end the command batch
batch end

# 
'''

[image]

etracer65, 2019-04-12:
I can't replicate this behavior. Please reset your flight controller to `defaults` and test. It's likely that one of the settings you changed to enable the RPM filter is interfering with the gyro signal lines.

mmakam2, 2019-04-12:
No, we figured it out in the slack channel, all the DMA or Timer lines were not in the diff.

mmakam2, 2019-04-12:
you have to use the snippet not the diff

etracer65, 2019-04-12:
Then can you close this and open an issue specifically about what might be missing from the `diff` and how to replicate?