betaflight/betaflight #7991: Kakute f7 65A stack - ESC issue

Tapwoot, 2019-04-12:
**Describe the bug**
The FC reads the ESC very random. Sometimes not at all, other times, 2 or 3 escs in the stack.

**To Reproduce**
Updated to betaflight 4.0

**Expected behavior**
I expected all 4 escs to work.

**Flight controller configuration**
# dump

# version
# Betaflight / KAKUTEF7 (KTF7) 4.0.0 Apr 11 2019 / 07:20:40 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name KAKUTEF7
manufacturer_id 

# name
name -

# resources
resource BEEPER 1 D15
resource MOTOR 1 B01
resource MOTOR 2 B00
resource MOTOR 3 E11
resource MOTOR 4 E09
resource MOTOR 5 NONE
resource MOTOR 6 A03
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 E13
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 D12
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 D05
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_TX 12 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 D06
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
[... 887 lines cut ...]
set pitch_expo = 38
set yaw_expo = 38
set roll_srate = 63
set pitch_srate = 63
set yaw_srate = 63
set tpa_rate = 50
set tpa_breakpoint = 1500
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

# 
**Setup / Versions**
 - Kakute f7
 - Im running latest Heli32 
 - By wired connectors.

**Additional context**
I downgraded to 3.5.7, and Im flying again.

McGiverGim, 2019-04-12:
Can you paste the output of a diff and not a dump command? Too much information to search in.

Tapwoot, 2019-04-12:
No, sry, its to late since Im back at 3.5.7. But I tried with stock motor resource, and it had the same result

etracer65, 2019-04-12:
If you can't provide the requested information there's little chance that we can do anything to diagnose or fix and we'll be forced to close this issue. It's unlikely that any dev has the same combination of hardware to replicate the problem.

Tapwoot, 2019-04-12:
Im gonna do a reflash of latest target withouth doing nothing, and se what happends. Thank you :)

closed 2019-05-19 by stale[bot]