betaflight/betaflight #7997: Unexpected behaviour on BF 4.0

fOmey, 2019-04-13:
**Describe the bug**
-
Drone shot up into the air while doing a hover test (captured blackbox data), the previous test resulted in the drone flipping out of the sky.

**Expected behavior**
-
Hover.

**Flight controller configuration**
-
```
# version
# Betaflight / FURYF4OSD (FY4O) 4.0.0 Apr 11 2019 / 07:16:52 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name FURYF4OSD
manufacturer_id 

# name

# resources
resource PPM 1 NONE
resource SERIAL_TX 11 C09
resource ESCSERIAL 1 NONE

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 2 8192 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1700 2100 0 0
aux 2 2 1 1300 1700 0 0
aux 3 13 3 1300 2100 0 0
aux 4 28 0 1700 2100 0 0
aux 5 35 2 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set rssi_channel = 9
set motor_pwm_protocol = DSHOT600
set bat_capacity = 850
set current_meter = ADC
set vbat_scale = 106
set ibata_scale = 275
set small_angle = 60
set pid_process_denom = 1
set osd_vbat_pos = 2519
set osd_rssi_pos = 2509
set osd_tim_1_pos = 2465
set osd_tim_2_pos = 2487
set osd_vtx_channel_pos = 2476
set osd_crosshairs_pos = 237
set osd_current_pos = 385
set osd_mah_drawn_pos = 449
set osd_avg_cell_voltage_pos = 2497
set vtx_channel = 6
set vtx_power = 3
set vtx_freq = 5840

# profile
profile 0

set vbat_pid_gain = ON

# rateprofile
rateprofile 0

# end the command batch
batch end
```

**Setup / Versions**
-
Diatone R349 - Mamba F4 Mini stack.

**Additional context**
-
Blackbox data: [btfl_001.zip](https://github.com/betaflight/betaflight/files/3075518/btfl_001.zip)

etracer65, 2019-04-13:
You may have too much vibration coming from your motors. Try reducing D on roll/pitch to around 27 and test.

closed 2019-04-13 by fOmey

fOmey, 2019-04-13:
This was covered in tuning tips, which I normally don't read as I always fly defaults.. I've followed the guide, will test it out on the field after I patch and paint the ceiling haha.