betaflight/betaflight #7998: Beta

btekfpv, 2019-04-13:
**Describe the bug**
A clear and concise description of what the bug is. This is a kiss clone flight controller, I had flown this board with v 3.5. Today I flashed the latest version 4.0, after the flash was complete, I disconnected from the USB port. Plugged back in to open the conf. On the set up screen the accelerometer showed my craft upside down. Clicking on the calibration button brings the craft back to upright but when I pick up the craft the roll axis and pitch axis are backwards. I go onto the configuration page and find that the roll is 180 degrees off and the yaw needs to be 90 rotated even though it shows 0. After correcting the values on the configuration page the craft responds correctly. However when on the motors tab, when plugging in the flight battery all chimes play out, but the motors will not spin but one. And it does not respond the throttle position. The motor number one in the sequence, spins at half throttle with the slide at zero and fluctuates with throttle movement to full throttle which is spinning the motor at 10% of the rpm range.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.
This flight controller has a DALrc 4-1 esc, a spektrum serial race receiver, TBS unify pro hv vtx, runcam micro camera, 2207/2450 kv motors
**Additional context**
Add any other context about the problem here.

btekfpv, 2019-04-13:
Sorry to add but, after it chime up, when the throttle is lifted one chime is heard then the motor spins without control

Docteh, 2019-04-13:
Hi, that sounds like a combination of two problems. ESC protocol, and possibly using the wrong target. I think you should fill in some more of the details that the template asks you to.

btekfpv, 2019-04-13:
Ok I did a diff and I am sorry I don’t know what a code block is. The wrong target would allow it to fly.

Docteh, 2019-04-13:
Just above the green Comment button there is a M with an arrow pointing down, its a link that explains markdown. They're talking about backtick characters.

\`\`\` See the cruft on either side of this line? \`\`\`

 those are the three backticks that they're talking about

closed 2019-04-14 by mikeller