betaflight/betaflight #7999: Feature Request: Add OSD profile auto-switch on failsafe

Xaverix, 2019-04-13:
With the new OSD profiles I can have my huge GPS data on one profile and other data used while flying on second profile.

Can You add a feature to automatically force selected OSD profile when betaflight detects failsafe?
This can be very helpful if I have GPS but I'm not using GPS rescue or GPS rescue will fail.
Seeing the GPS data even for a while might help finding the quad.

Docteh, 2019-04-13:
How many AUX channels are you using? If you're not using all 14 AUX channels you could set eg AUX14 to a position that isn't 1500 as a failsafe...

pkruger, 2019-04-27:
What about displaying the GPS Lat and Lng when disarming, i.e. as part of the stats screen, would that help?

closed 2026-09-26 by blckmn