betaflight/betaflight #8001: OSD Warnings for RSSI and Link Quality, also Link Quality alarm

ianrmurphy, 2019-04-13:
For discussion in advance of issuing a Pull Request:

I've implemented OSD Warnings for RSSI and Link Quality, and added an alarm setting for Link Quality. See https://github.com/ianrmurphy/betaflight/tree/add-rssi-lq-osd-warnings (and https://github.com/ianrmurphy/betaflight-configurator/tree/add-rssi-lq-osd-warnings for corresponding configurator updates). Note that the target code depends on https://github.com/betaflight/betaflight/pull/7968. Has received preliminary static testing on 2 targets with `#define USE_RX_LINK_QUALITY_INFO` but not without.

Rationale:
Previously low RSSI alarm would only cause the RSSI OSD field to flash. 1) this may not be set up on all OSDs (by preference or omission) and 2) this may be missed as a small number likely at the edge of the field of view. 

Compare this to the 'Battery Warning' and 'Battery Critical' warnings which appear flashing in the centre of the OSD and therefore command pilot attention. They appear even if the voltage(s) are not displayed in the normal OSD. I consider RSSI just as important as battery voltage for safe flying, especially long range.

Finally, if this rationale applies to RSSI then it should also apply to Link Quality, so I have implemented this also. This requires adding an alarm threshold for Link Quality.

I think the following needs to be discussed before making a Pull Request:
1. Heirarchy of Warnings - currently RSSI and Link Quality are at the end of the list, putting them lowest in the heirarchy. Battery Warnings are near the top of the list. Should RSSI & LQ be moved up?
2. Default settings - I have chosen 80% as the default LQ alarm threshold, but have no real data to back this up. Also, I have defaulted these warnings to OFF, unlike all the others, to avoid setups without RSSI or LQ having a constant warning. Should these defaults be adjusted? Also, should a setup without RSSI measurement default to 100% RSSI instead of 0%?
3. CLI settings & MSP API integration - are these correct / adequate?

mikeller, 2019-04-14:
@ianrmurphy: Sounds like a reasonable feature to be added.

> Heirarchy of Warnings - currently RSSI and Link Quality are at the end of the list, putting them lowest in the heirarchy. Battery Warnings are near the top of the list. Should RSSI & LQ be moved up?

Yes, I think they should be even above the battery warning - flat battery or not, a craft will potentially be lost if it flies out of range.

> Default settings - I have chosen 80% as the default LQ alarm threshold, but have no real data to back this up. Also, I have defaulted these warnings to OFF, unlike all the others, to avoid setups without RSSI or LQ having a constant warning. Should these defaults be adjusted?

Sounds reasonable. Defaulting to off is probably right, as this will need tuning of the minimum to avoid just being an annoyance and resulting in the pilot ignoring warnings.

> Also, should a setup without RSSI measurement default to 100% RSSI instead of 0%?

I wouldn't - with 100% there's the chance that if the something stops working a pilot will be lulled into a feeling of false security.

CLI settings & MSP API integration - are these correct / adequate?

Open a pull request and we'll see.

laz2wiringpi, 2019-05-06:
@ianrmurphy should this be closed it was merged ?
or is there something more to the issue that was not dealt with?

ianrmurphy, 2019-05-06:
Yes. I'll close it now...

closed 2019-05-06 by ianrmurphy