betaflight/betaflight #8002: Matek F722 Mini ADC Not working

nikolaswidell, 2019-04-13:
**Describe the bug**
I get 2.2A current without motors spinning, in the air I get around 2-20A values which are somehow related to the throttle position.

I got correct scales for my 4in1 ESC analog current sensor. Works flawlessly on 3.5.0. Does not work on 3.5.7 ja 4.0 though.

I just noticed that if I drop to 8/2k loop rates the current sensor works like it worked on 3.5.0. Switching to 8/8 it fails again. On both 3.5.7 and 4.0.

**To Reproduce**
Just plug battery and run 8/8

**Expected behavior**
Correct current readings

**Flight controller configuration**
```
 diff

# version
# Betaflight / MATEKF722SE (MF7S) 4.0.0 Apr 11 2019 / 07:26:45 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id 

# name
name gm

# resources
resource CAMERA_CONTROL 1 A01

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature ESC_SENSOR

# beeper

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# led
led 0 0,0::CT:1
led 1 1,0::CT:1
led 2 2,0::CT:1
led 3 3,0::CT:1
led 4 4,0::CT:1
led 5 5,0::CT:1

# color
color 0 0,0,255
color 1 117,0,206
color 2 238,9,255
color 3 359,255,255

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 1 1700 2100 0 0
aux 2 35 1 1700 2100 0 0
aux 3 39 2 1700 2100 0 0
aux 4 49 2 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_scale = 200
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set min_throttle = 1020
set dshot_idle_value = 350
set motor_pwm_protocol = DSHOT1200
set vbat_max_cell_voltage = 450
set vbat_warning_cell_voltage = 340
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set runaway_takeoff_prevention = OFF
set transient_throttle_limit = 10
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 40
set osd_cap_alarm = 1200
set osd_tim1 = 1
set osd_tim2 = 0
set osd_vbat_pos = 449
set osd_rssi_pos = 2080
set osd_tim_1_pos = 2487
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_g_force_pos = 271
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 0
set osd_ah_pos = 200
set osd_current_pos = 2433
set osd_mah_drawn_pos = 2457
set osd_craft_name_pos = 2075
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 344
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 0
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2442
set osd_avg_cell_voltage_pos = 2464
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 330
set osd_disarmed_pos = 2412
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 2068
set osd_esc_rpm_pos = 68
set osd_core_temp_pos = 2072
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set vtx_band = 5
set vtx_channel = 8
set vtx_freq = 5917

# profile
profile 0

set iterm_relax_cutoff = 35
set i_pitch = 85
set d_pitch = 28
set i_roll = 80
set d_roll = 30
set p_yaw = 30
set i_yaw = 90
set f_yaw = 30
set d_min_roll = 16
set d_min_pitch = 18

# rateprofile
rateprofile 0

set roll_rc_rate = 167
set pitch_rc_rate = 167
set yaw_rc_rate = 167
set roll_expo = 2
set pitch_expo = 2
set yaw_expo = 2
set tpa_rate = 75
set tpa_breakpoint = 1400
```
**Setup / Versions**
 - Matek F722-Mini, banggood
 - Spedix GS40A, R9 Mini, TBS Unify Nano

**Additional context**
I've read about something like this in this repo but can't find the issues. Had something to do about DMA or RAM or etc. Also included matek F7 board.

laz2wiringpi, 2019-04-13:
resource ADC_CURR 1 C01
dma ADC 1 0
set current_meter = ADC
set battery_meter = ADC

try the unified .
i have a MATEKF722SE not mini and it works ..

nikolaswidell, 2019-04-13:
Tried it, not working :( 

> # dma ADC 1 0
> dma ADC 1: no change: 0

Got this though?

closed 2022-05-10 by haslinghuis