betaflight/betaflight #8003: BF 4.0.0 for CrazybeeF3FR not supporting external FS RX on UART3

maylibu, 2019-04-13:
On 4.0.0, I cannot get Betaflight to detect my external FS rx wired on uart3 on my CRAZYBEEF3 board with onboard FR rx. This works fine on 3.5.7. 

I have the Crazybee F3 FR that came with my Snapper7. I have connected a Micro IA8X (flysky rx that came with the UFPV Nirvana) on UART3.

To reproduce, flash CrazybeeF3FR 4.0.0 with full settings erase, connect, set port uart3 to serial rx, save/reboot, set configuration receiver-mode to serial-based and provider to IBUS, save/reboot. Check receiver tab to see no input. Reset to check settings saved, still no input. Rebind, still no input.

Expected behavior: (I did this *after* the above to confirm I don't have a HW issue.)
Same steps as above except on 3.5.7. Receiver tab shows inputs from my tx.

I assume a diff is not needed since I am starting from scratch/default.

McGiverGim, 2019-04-13:
Where is the UART3 located? I can't find it in my CRAZYBEEF3FR.

maylibu, 2019-04-13:
UART3 is on the underside of theboard slightly left of the gnd/5v/vin/vout pads. See http://img.banggood.com/file/products/20180629031334HowtouseExternalrecevieronCrazybeeF3flightcontroller.pdf

It looks like this may have been added on the more recent version of the board.

McGiverGim, 2019-04-14:
Thanks! Mine must be older, it does not have this pads.

mikeller, 2019-04-14:
@maylibu: The CRAZYBEEF3 boards are built without iBus support: 

https://github.com/betaflight/betaflight/blob/9ad2cc84452b8dcfa385bc5d8ea3cf9ab86ad046/src/main/target/CRAZYBEEF3FR/target.h#L58

This means connecting an external iBus RX will not work unless you use a custom built version of the firmware.
The reason for this is that these boards did not have an accessible UART until recent versions, and the manufacturer neglected to update their targets to reflect this change.

huangdequan75, 2019-04-14:
Betaflight 10.5.1 does not show the OSD and Mode tabs, it shows an empty pane and then freezes. 10.4 works fine for CrazybeeF3FR. Good thing i have the old version on another notebook.

McGiverGim, 2019-04-14:
That's not true, you have an unofficial/beta version of Betaflight 4.0. Update to the official version or ask the manufacturer of your quad.

maylibu, 2019-04-14:
Do I need to make a custom build or is this something that will be available in a future 4.0.x release?

mikeller, 2019-04-14:
@maylibu: You will have to use a custom build. The serial RX drivers were removed to save space and make it possible to support other features in the freed up space. The assumption is that most users will want to use these boards with the RX that comes on board, and get more benefit out of the other features that were enabled as a trade-off.

maylibu, 2019-04-15:
To make a custom build, will I have to do more than remove/comment the line 
#undef USE_SERIALRX_IBUS ?
Do I have to disable some other feature? Is there a recommended feature that is recommended to remove to give the space needed, but is not too useful on a brushless micro quad?

tloki, 2019-04-15:
I thing you should look here:

https://github.com/betaflight/betaflight/blob/master/docs/Customized%20Version.md#specific-features-for-each-flight-controller

Following those instructions, I have built custom 4.0 for my CC3D F1 board without problems (1.6kHz PID loop, spektrum telemetry, RSSI and Voltage sensing).

The only thing you should be edditing is `target/[FLIGHT_CONTROLLER_NAME]/target.h` and build it using super-easy docker container available here:

https://github.com/betaflight/docker-betaflight-build

If you're on Windows, I think the simplest option is just to install Virtualbox and use some pre-installed Linux (say Ubuntu from osboxes.org) image.

mikeller, 2019-04-15:
@maylibu: You'll find out soon if your build runs out of flash space - it will fail. If you need to disable something, I'd start with `USE_RX_SPI`, as you won't be needing that if you're going to be using an external RX.

maylibu, 2019-04-15:
Thanks, I will check it out!

closed 2019-05-22 by stale[bot]

rmarcoro, 2020-10-02:
Hi all
I have the same issue.
UR65 CRAZYBEEF3FS FC with internal SPI Flysky receiver and Betaflight 4.0.
It bind ok with my ir8x Trasmitter using IBUS and AFHDS protocol. 
I'd like to bind the quad to my newest Taranis X9 lite using an ARCHER M+ receiver soldered on FC RX3 pad.
I soldered the receiver, set up serial receiver on UART3  and Serial receiver SBUS.
No success on the 1st try. FC does not recognizes the receiver.
After reading this page I think... what happens if I update to Firmware FR version?
I do this and it works !!!!!!!!
So I have a FC with internal SPI Flysky receiver flashed with a Frysky version Firmware.
Now I can bind a FS board with Betaflight 4.0 with a external FR receiver to my Taranis trasmitter.
Tks all. This information above help me to have this idea.

McGiverGim, 2020-10-02:
The space is very limited in F3 so the fs version does not contain the frsky driver. You can build a custom firmware if you want adding it.

rmarcoro, 2020-10-02:
Ok, @McGiverGim, as I see above on this thread.
As I told. You can just flash the FC with a Frysky Betaflight Firmware. Even your FC is a Flysky version.
Do this only if you want to use a external receiver.

rdl99, 2021-09-27:
> On 4.0.0, I cannot get Betaflight to detect my external FS rx wired on uart3 on my CRAZYBEEF3 board with onboard FR rx. This works fine on 3.5.7.
>
> I have the Crazybee F3 FR that came with my [...]

thank for shearing.....i run into the same problem on my US65 CRAZYBEEF3FR.....I dont have an fr tx, so im trying to get an ORX tx with a serial speketrum rx to work. luckely it has a uart 3 pad