betaflight/betaflight #8004: LED Conflict RPM filter

Gaz-B, 2019-04-13:
This seems very odd but if I have LEDs enabled with RPM filter LEDs don't work and motors spin for a fraction of a second then shut off with LEDs switched of in the configurator motors work fine. This isn't a new build LEDs function is fine up until I add the RPM Filter changes. This is operating motors from Betaflight or BL-32 configurators. Pyrodrone F4 and Typhoon v2.? 4 in 1. full release 4.0 and BL32.6.8.

Gaz-B, 2019-04-13:
Good in that case I shall ignore it. I only mentioned the configurator to point out I was testing the motor function from my PC and not the RX.

MeatG, 2019-04-20:
I have the same problem with Pyrodrone F4 v2. Maybe there is a workaround with resource remapping?

Gaz-B, 2019-04-21:
I found there was another issue with mine so have given up on the filtering for now. I'm using pyrodrone F4 boards and one of the tested ones apparently and my video randomly goes when throttle is used but still have OSD. I replaced the camera cable then the camera didn't help then reflashed without activating the new filters and it fixed it.

Gaz-B, 2019-04-21:
I'm now even more confused as I had only flown this machine once since taking the filters off and it went well this was in the dark over a car park and this isn't a car park basher. I now went to do more testing tuning and the video goes black every time I arm just gone back to 3.5.7 and dumped my last full diff back in and I can now arm a throttle with clean video ?. I feel very lucky not to have smashed it into the asphalt in the dark.

MeatG, 2019-04-21:
Note: Bi-directional RPM filter works when you disable LEDs by typing resource LED_STRIP 1 NONE in the Betaflight Configurator CLI, even if you don't have any LEDs installed or configured.

MeatG, 2019-05-24:
I think this is still an issue