betaflight/betaflight #8008: Exclamation mark (!) missing in OSD font

TonyBlit, 2019-04-14:
The exclamation mark (!) is missing on the OSD font. In its place (0x21) there's a small '100'. So when the quad goes into failsafe, the flight mode shows 100FS100 instead of !FS!
I've been observing this problem for a good amount of months while tinkering with GPS Rescue on earlier BF4 builds, and I assumed it was because I didn't have the latest OSD font. But now that the new BF Configurator is out, I've flashed the latest OSD Font with 10.5.1 and I still observe the same problem.
If the exclamation mark has been permanently removed, then !FS! can be replaced with -FS- or whatever. I can do the change myself, but I wanted to raise the issue first, just in case.

mikeller, 2019-04-14:
Lol. That's pretty rubbish100

My hunch is that this should be fixed by reinstating '!' as 0x21. But since this will require for everybody to re-flash their font this will be a more long term solution.

Replacing the exclamation mark with a '-' or '*' will do as a short term fix.

Docteh, 2019-04-14:
Is anything using 0x21 as 100? in the font there are the following symbols in sequence: 100, 500, 2k5 MAX 

Looks like the fonts had those symbols the whole time.

mikeller, 2019-04-14:
@Docteh: All access for OSD special characters (i.e. not ones that correspond to ASCII characters) should only be through defines in `src/main/drivers/max7456_symbols.h` (and conversely, there should not be any defines in there for symbols that are not used).
This works well except for symbols that we are using as printable ASCII characters.

TonyBlit, 2019-04-15:
Actually, in max7456_symbols.h 0x21 is not defined, so maybe @Docteh is right and the 100, 200, 2k5, MAX symbols are not used after all?

Docteh, 2019-04-15:
I still need to read through all of `src/main/osd/osd_elements.c` But I'm wondering if it'd be a good idea to do a general cleanup of the OSD fonts. I checked fonts back to 2016-06-29 and backtracked some of the OSD code. I'm surprised nobody pointed out the exclaimation mark being wrong. Before it was "!FS!" it was... "!FS" 

0x21-0x24 could be replaced, and the other one I'm really curious about is 0x3F (Question Mark) I figured for sure the CMS OSD would ask a user something like "Are you sure?" But from 2016 til now 0x3F is just a blank space in the fonts. 

0x89 0x98 0x9D 0x9E I didn't look back in the code for these ones, but they seem pretty orphaned to me. Unless GY+ means something.  

[image]

iNav has some exclaimation marks we could copy. Although after looking at them I'd be very tempted to redo/tweak them all except for the one in their default.mcm, that one is fine. 

@mikeller  I'll look into it drawing some characters this week, unless someone waves me off.

mikeller, 2019-04-15:
@TonyBlit: That was the point of my statement - sorry for not being very clear about this.

@Docteh: As I said above, all characters that don't have a define in `src/main/drivers/max7456_symbols.h` (and aren't used as ASCII characters) are up for grabs. In the past we have not removed unused characters from the font (as this makes no difference to the user), but I would be all for restoring ASCII characters that we actually have use for, like '!' and '?'.

closed 2019-04-19 by mikeller