betaflight/betaflight #8011: Hi

drdrechronik, 2019-04-14:
**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

dadoribeiro, 2019-04-15:
Did you load the snippet for timer and DMA changes?

drdrechronik, 2019-04-15:
Thanks for the response.
I may have forgotten it earlier. I just tried it again with the snippet loaded, the two last beeps still don't sound....
Here are the settings I tried it with:

# diff

# version
# Betaflight / MATEKF722SE (MF7S) 4.0.0 Apr 11 2019 / 07:26:45 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id 

# name

# resources
resource MOTOR 1 B01
resource MOTOR 2 B00
resource MOTOR 3 B05
resource MOTOR 4 B04
resource CAMERA_CONTROL 1 A01

# timer
timer A03 1

# dma
dma pin B04 NONE
dma pin B05 NONE
dma pin B00 NONE
dma pin B01 NONE
dma pin A15 NONE
dma pin B03 NONE
dma pin B06 NONE
dma pin B07 NONE
dma pin A08 NONE
dma pin A03 0
# pin A03: DMA1 Stream 1 Channel 6
dma pin A01 NONE
dma pin A00 NONE

# mixer

# servo

# servo mix

# feature
feature GPS
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 2 115200 9600 0 115200
serial 2 8192 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1900 2100 0 0
aux 1 46 1 900 1100 0 0
aux 2 13 1 1900 2100 0 0
aux 3 35 1 1700 1900 0 0
aux 4 39 1 1550 1700 0 0
aux 5 40 1 1300 1450 0 0
aux 6 49 1 1100 1300 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 0
set gyro_lowpass2_type = PT1
set gyro_lowpass2_hz = 150
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = -36,122,309
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_derivative_hz = 150
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = PT1
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set dshot_idle_value = 350
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set align_board_yaw = 270
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_ublox_use_galileo = ON
set gps_rescue_angle = 40
set gps_rescue_initial_alt = 30
set gps_rescue_sanity_checks = RESCUE_SANITY_FS_ONLY
set gps_rescue_min_sats = 5
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 1
set osd_warn_no_gps_rescue = OFF
set osd_cap_alarm = 1000
set osd_alt_alarm = 50
set osd_vbat_pos = 2487
set osd_rssi_pos = 2477
set osd_link_quality_pos = 430
set osd_tim_1_pos = 2508
set osd_tim_2_pos = 33
set osd_flymode_pos = 35
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 2466
set osd_mah_drawn_pos = 2498
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 2102
set osd_home_dir_pos = 2092
set osd_home_dist_pos = 2095
set osd_compass_bar_pos = 266
set osd_altitude_pos = 2083
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2519
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_max_alt = ON
set osd_stat_max_fft = ON
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_channel = 4
set vtx_freq = 5800

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 7000
set feedforward_transition = 10
set p_pitch = 55
set d_pitch = 30
set p_roll = 45
set d_roll = 30
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# rateprofile
rateprofile 0

set roll_rc_rate = 108
set pitch_rc_rate = 108
set yaw_rc_rate = 120
set roll_expo = 12
set pitch_expo = 12
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 75

# end the command batch
batch end

#

drdrechronik, 2019-04-15:
So, I'm one step further, I disabled Blackbox, since "# SPI 3 TX for SDCARD conflicts witn M2" for my target.

so I changed:

set blackbox_p_ratio = 0

and added:

set blackbox_device = NONE

to the diff above. 

Now I get a response when typing "status", here it is:

# status
MCU F722 Clock=216MHz, Vref=3.32V, Core temp=49degC
Stack size: 2048, Stack address: 0x20010000
Config size: 3043, Max available config: 16384
GYRO=MPU6000
System Uptime: 15 seconds, Current Time: 2019-04-15T08:57:59.286+00:00
CPU:4%, cycle time: 251, GYRO rate: 3984, RX rate: 33, System rate: 9
Voltage: 1508 * 0.01V (4S battery - OK)
I2C Errors: 0
SD card: Startup failed
Arming disable flags: RXLOSS CLI MSP RPMFILTER
Dshot reads: 0
Dshot invalid pkts: 0
Dshot irq micros: 0
Dshot RPM Motor 0: 0
Dshot RPM Motor 1: 0
Dshot RPM Motor 2: 0
Dshot RPM Motor 3: 0
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 

Still beeps missing.

dadoribeiro, 2019-04-15:
If you disable dshot_bidir it arms ok? Can you connect to BLHeli32 with bidir enabled? Disabled? Check if all ESCs are really with the same firmware.