betaflight/betaflight #8018: OSD element positions not unique between Profiles

Phrozentech, 2019-04-14:
Activating any element for the OSD, it's position is not unique across profiles and remains static.
I would expect this type of behavior, because it is not called out in the configurator that elements retain their position across all profiles and this feature simply allows a user to turn on/off elements for each profile.

**Flight controller configuration**
```
# diff

# version
# Betaflight / MATEKF722SE (MF7S) 4.0.0 Apr  9 2019 / 10:31:52 (2b79f2b1a) MSP API: 1.41

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id 

# name
name LongRangeStyle

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature GPS
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon

# map
map TAER1234

# serial
serial 0 2 115200 57600 0 115200
serial 2 2048 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1500 2100 0 0
aux 1 27 1 1700 2100 0 0
aux 2 46 1 1300 1700 0 0
aux 3 13 1 1800 2100 0 0
aux 4 35 2 1700 2100 0 0

# adjrange
adjrange 0 0 3 900 2100 29 3 0 0

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set acc_calibration = -3,-21,338
set mag_calibration = 229,174,1
set baro_hardware = NONE
set rssi_channel = 9
set fpv_mix_degrees = 20
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set motor_pwm_protocol = DSHOT600
set failsafe_procedure = GPS-RESCUE
set vbat_max_cell_voltage = 435
set current_meter = ESC
set battery_meter = ESC
set ibata_scale = 175
set yaw_motors_reversed = ON
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set pid_process_denom = 2
set osd_warn_batt_not_full = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 40
set osd_cap_alarm = 1300
set osd_vbat_pos = 2049
set osd_rssi_pos = 14362
set osd_tim_2_pos = 2455
set osd_throttle_pos = 2104
set osd_vtx_channel_pos = 14337
set osd_mah_drawn_pos = 14722
set osd_gps_lon_pos = 4496
set osd_gps_lat_pos = 4464
set osd_home_dir_pos = 4109
set osd_home_dist_pos = 4111
set osd_compass_bar_pos = 204
set osd_avg_cell_voltage_pos = 6497
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set vtx_band = 5
set vtx_freq = 5658
set vcd_video_system = NTSC

# profile
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 130
set pitch_rc_rate = 130
set yaw_rc_rate = 155
set roll_expo = 5
set pitch_expo = 5
set yaw_expo = 10
set yaw_srate = 75

# end the command batch
batch end
```

mikeller, 2019-04-14:
@Phrozentech: Can you please point out what part of the documentation for osd profiles (https://github.com/betaflight/betaflight/blob/master/docs/OSD%20Profiles.md) is not clear enough and needs to be amended?

Phrozentech, 2019-04-15:
> 
> 
> @Phrozentech: Can you please point out what part of the documentation for osd profiles (https://github.com/betaflight/betaflight/blob/master/docs/OSD%20Profiles.md) is not clear enough and needs to be amended?

Absolutely correct i didn't even check the Wiki my apologize. I was expecting to see something in the configurator mentioning this.

closed 2019-04-15 by Phrozentech