betaflight/betaflight #8021: No movement in receiver tab DSMX

fletch3d, 2019-04-15:
Have an Eachine Aurora 100 with updated Minicube F4 FC running DSMX sat.  Worked on prior versions.  Re-bound the DSMX after update to 4.0 without issue, but nothing is moving in the Receiver tab.  Omnibusf4 firmware.  DSMX is on Uart1, all good.  Have a solid bind yellow on receiver, have a blue blinking light on FC (I think that is for Frsky reference).  Here is my Diff:

# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Apr 11 2019 / 07:31:36 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id 

# name
name Fletch Aurora

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature SOFTSERIAL
feature LED_STRIP

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE

# beacon

# map
map TAER1234

# serial
serial 0 64 115200 57600 0 115200

# led
led 0 0,7::COI:0
led 1 7,7::CTO:0
led 2 8,7::CTO:0
led 3 15,7::COI:0

# color

# mode_color

# aux
aux 0 0 1 900 1275 0 0
aux 1 1 0 900 1300 0 0
aux 2 2 0 1300 1700 0 0
aux 3 4 1 900 1300 0 0
aux 4 13 2 1550 2100 0 0
aux 5 35 3 900 1700 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set baro_hardware = NONE
set serialrx_provider = SPEK2048
set motor_pwm_protocol = DSHOT600
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_cap_alarm = 550
set osd_tim1 = 768
set osd_vbat_pos = 2432
set osd_rssi_pos = 2426
set osd_tim_1_pos = 2368
set osd_tim_2_pos = 375
set osd_flymode_pos = 2458
set osd_throttle_pos = 322
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 2400
set osd_craft_name_pos = 2440
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 2408
set osd_disarmed_pos = 2252
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83

# profile
profile 0

# rateprofile
rateprofile 0

watageek, 2019-04-15:
Have a look maybe it will soved your issue.

https://www.rcgroups.com/forums/showthread.php?3271171-Betaflight-4-0-Omnibus-F4-Pro-V2-Serial_RX-DSMX-not-working-on-UART1-_-FIXED

andrejpodzimek, 2019-04-25:
I think I have the same problem on [this Eachine Minicube](https://www.banggood.com/Eachine-Minicube-20x20mm-F4-OSD-Compatible-Frsky-Flysky-DSM-RX-Blheli_S-10A-For-Aurora-68-90-100-p-1165366.html). It's the DSMX version, everything worked fine in Betaflight 3.3.3, but there's no movement in the receiver tab in Betaflight 4.0.0. (The transmitter is a Walkera Devo 10.) The receiver is bound and has a steady light. I tried to rebind it a few times in both 11ms ("fast") and 22ms DSMX modes, but no luck; there seems to be no serial communication on UART1. The receiver binding seems to work though (I mean, initiated from the flight controller with `spektrum_bind`).

andrejpodzimek, 2019-04-25:
As @jflyper says in #8094, reverting 7665a424c7386ed90f7bd042e265a845a41fd63e solves this for me. I have 4.0.0 with just that change reverted and the DSMX receiver works again.

watageek, 2019-05-07:
Hi @fletch3d
If you can check if there is still an issue with the official 4.0.2 release. 
I have posted the " temporary workaround" on RC Group. If there is no more issue with the 4.0.x maintenance branch, I will stop posting the updated fix.

mikeller, 2019-05-07:
@watageek: Can you please comply with the license of Betaflight, and post the source code used to build the binaries you are publishing?

Also, you should maybe consider opening a pull request with the fix that you have found, so it can be discussed.

fletch3d, 2019-05-07:
> Hi @fletch3d
> If you can check if there is still an issue with the official 4.0.2 release.
> I have posted the " temporary workaround" on RC Group. If there is no more issue with the 4.0.x maintenance branch, I will stop posting the updated fix.

Hey watageek,

Will test in 4.0.2.  will report back.

watageek, 2019-05-07:
@mikeller
Sorry I will.

watageek, 2019-05-07:
@fletch3d
@mikeller
As soon as you give me feed back, I will do the branch. Hope for the best.

jflyper, 2019-05-08:
Just a reminder.

There is an issue ticket #8094 that points to the most probable cause for this issue, and I have already requested someone with the actual non-working hardware and equipment to do some in depth analysis.

MJ666, 2019-05-08:
I have seen an similar issue with 4.0.2 on an AFNG F7 and an LemonRX DSMX satelite. There is no RX ressponse at all in 4.0.2. The TX is an Walkera Devo10 with Deviation firmware. BF 4.0.0 and actual master fom yesterday is working fine. Not sure if this is related.

mikeller, 2019-05-08:
@MJ666: The number of changesets between 4.0.0 and 4.0.2 is relatively small, can I get you to bisect to find what changeset is breaking in your reproduction?

MJ666, 2019-05-09:
@mikeller will be out next few days. Will give it a try when back and have acess to my gear.

mikeller, 2019-05-11:
@watageek: Removing your posts peddling your 'workaround' until you comply with the license for Betaflight.

watageek, 2019-05-11:
@mikeller

Sorry... again...

I will punish myself sitting quietly in a corner for 15 minutes, Then go fly, then do the fork.

One of my ex-boss told me one time , I learn fast but he need to explain often. 😐

watageek, 2019-05-11:
@mikeller

Sorry... again...

Is it ok for you ( betaflight code of conduct) if I just put a .zip with the conditional update ( i.e. 'The workaround') attached?

As jflyper quote already revert to old commit is not the way to go. 

Do you want me to post the .zip at rc group too?

mikeller, 2019-05-11:
@watageek: If that is all that the workaround is then yes - this isn't something that will be workable as a solution, so there is not much value in opening a pull request. I'd probably also add an explanation that all you've done is removed a fix, and that users should expect to get problems with other devices if they are using your version. According to the GPL you should post it to the same place as where you are posting the binaries - which is RCGroups.

MJ666, 2019-05-18:
Hello after retesting with various different AFNG F7 FC's i could not reproduce the problem i had initially with 4.0.2? May be it was an bad flash or some other issue. i had flashed 4.0.2 alredy 2 times with the bad result at that time? All FC's including the one i had problems are working now well with an DSMX LemonRX and 4.0.2.

mikeller, 2019-05-18:
Thanks @MJ666. This looks more and more like yet another issue of a bad clone.

@fletch3d: Can you try if setting the following in CLI solves the problem for you:

```
resource pinio 1 b8
set pinio_config = 1,1,1,1
```

MikeNH, 2019-06-02:
FrSky RXSR-FC - Omnibus F4 V6 Flight Controller w/ Integrated R-XSR and Devo 7e radio with DeviationTX firmware. Betaflight target OMNIBUSF4V6.

Everything working well with Betaflight 3.5.5. I did a cli "dump" as a backup before upgrading to Betaflight 4.0.3. After the upgrade and configuration, 
 channels on Receiver tab were unresponsive.  Receiver and radio were bound properly.  
 
Re-flashed with 3.5.5 to confirm hardware was working correctly. 

I will look for the referenced workaround on RC Groups.

watageek, 2019-06-02:
@MikeNH

Have you tried this fix? 
'''resource pinio 1 b8
set pinio_config = 1,1,1,1'''

I have remove the fix until I have packed the code along with it. I am sorry I didn't have much spare time recently. And I would add It would be couter intuitive to go against the dev's software devellopment on this specific topic since it might be solely related to cheap clone board design

MikeNH, 2019-06-02:
Entering
      resource pinio 1 b8
resulted in 3 lines of text indicating b8 is alreaady assigned but did assign this command

set pinio_config = 1,1,1,1
succeeded.

Problem persists.

closed 2019-07-10 by stale[bot]

fletch3d, 2019-10-03:
Hey Guys,

Same issue in 4.0.6 and 4.1. No movement in betaflight receiver tab for this flight controller.

quiljw01, 2019-10-21:
I have this issue.  LemonRx receiver.  Obf4 airbot fc. BF 4.1.
Works fine on BF 3.5.7
Please help.

mikeller, 2019-10-21:
@quiljw01: This will potentially need a specific pull up / pull down on a pin. Maybe @flyinglemonfpv can help?