betaflight/betaflight #8023: GPS module not detected on power up

grahamcropley, 2019-04-15:
When quad is powered up from lipo, the GPS doesn't appear on the OSD. Replug and plug in, and it always fails to show up, this is reproduceable every time for me.

To get it to work, after the lipo is connected, I use the CMS/OSD Menu command to reboot the FC, then the GPS shows up on the OSD correctly - (I have DVR video to show if Im not clear how I'm explaining it)

I assume when it's all powered on together something is out with the timing or auto-detection or something? but when the FC is rebooted, the GPS is already powered on and 'running' so it is detected properly.

Flight Controller is Matek F722-SE
GPS Module is Matek GPS with Compass (M8Q-5883)
Betaflight 4 (release version)

I also have the compass wired up to SCL/SDA pins.

Other UARTs are used by Crossfire CRSF and TBS SmartAudio as well as GPS mentioned above.

grahamcropley, 2019-04-15:
diff all

```# diff all

# version
# Betaflight / MATEKF722SE (MF7S) 4.0.0 Apr 11 2019 / 07:26:45 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name MATEKF722SE
manufacturer_id 
mcu_id 003b00193237510c34363730
signature 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature GPS

# beeper

# beacon

# map
map TAER1234

# serial
serial 0 2 115200 115200 0 115200
serial 2 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 2 1300 1700 0 0
aux 2 46 2 1700 2100 0 0
aux 3 13 3 1700 2100 0 0
aux 4 26 1 1300 1700 0 0
aux 5 31 1 1700 2100 0 0
aux 6 36 3 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 11 s 1000

# display_name

# master
set dyn_lpf_gyro_max_hz = 500
set acc_calibration = 20,-36,197
set align_mag = CW90FLIP
set mag_declination = -1000
set mag_calibration = 55,120,452
set rssi_channel = 12
set fpv_mix_degrees = 25
set serialrx_provider = CRSF
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT1200
set small_angle = 180
set gps_provider = UBLOX
set gps_sbas_mode = EGNOS
set gps_set_home_point_once = ON
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 75
set osd_cap_alarm = 2000
set osd_alt_alarm = 400
set osd_vbat_pos = 2465
set osd_rssi_pos = 2137
set osd_tim_2_pos = 2518
set osd_current_pos = 2401
set osd_mah_drawn_pos = 2433
set osd_gps_speed_pos = 2145
set osd_gps_lon_pos = 2096
set osd_gps_lat_pos = 2081
set osd_gps_sats_pos = 2168
set osd_home_dir_pos = 2125
set osd_home_dist_pos = 2126
set osd_compass_bar_pos = 2154
set osd_altitude_pos = 2113
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2497
set osd_stat_rtc_date_time = ON
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_max_curr = OFF
set vtx_band = 5
set vtx_channel = 3
set vtx_power = 2
set vtx_freq = 5732

# profile
profile 0

# profile
profile 1

set dyn_lpf_dterm_min_hz = 120
set dyn_lpf_dterm_max_hz = 180
set dterm_lowpass_hz = 120
set p_pitch = 90
set i_pitch = 95
set d_pitch = 32
set f_pitch = 60
set p_roll = 90
set i_roll = 95
set d_roll = 32
set f_roll = 60
set i_yaw = 125
set f_yaw = 100

# profile
profile 2

# restore original profile selection
profile 1

# rateprofile
rateprofile 0

set roll_rc_rate = 110
set pitch_rc_rate = 110
set roll_srate = 78
set pitch_srate = 78

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0
```

McGiverGim, 2019-04-15:
Strange. I've the same FC but different GPS and it seems to work for me. 
When connected to the Configurator, the top GPS icon lights or it needs the reboot too?

grahamcropley, 2019-04-17:
Sorry for the delay in responding.

Running BF 3.5.7, the GPS is detected and displayed on the OSD when the lipo is connected.
Running BF 4.0.0, the GPS is not detected and not displayed on the OSD when the lipo is connected (until the FC is rebooted via stick commands - i.e. the GPS module is already on).

If I power up via lipo, and then connect USB.. the GPS icon is not on in BF Configurator. If I then click 'save and reboot' or use the stick commands.. GPS icon lights up.

So I think I've ruled out hardware or wiring issues. And seems like a start-up detection thing that has changed between 3.5.7 and 4.0.0

I've recorded DVR that I'll upload and link shortly.

grahamcropley, 2019-04-17:
Not that you didn't believe me... but still here's a repro of the issue - https://youtu.be/-Qf339Z6-24

IllusionFpv, 2019-04-19:
I've investigated this issue after it happened to me last time I've flew.
copy diff, reset settings and re-apply diff did worked.

grahamcropley, 2019-04-19:
Thanks for commenting @IllusionFpv - I did that many times to record the DVR to reproduce the issue. Going back and forth between the versions. I always have full chip erase, but I've also done it without re-pasting any diffs, just setting up GPS from scratch. It's the same experience regardless. 3.5.7 works first time.. 4.0.0 requires a reboot after it's powered up.

IllusionFpv, 2019-04-19:
Can you try to install 4.0.0-RC6 and test it?

grahamcropley, 2019-04-19:
Pretty sure it was on RC6 when I first noticed it, then final came out and it was the same.. but I will also try 3.5.7 > 4.0.0 RC1 to 6 and see where it changes.

grahamcropley, 2019-04-25:
Confirmed it also happens on RC1... so something changed between 3.5.7 and 4.0.0 RC1 to do with GPS detection or something?

aowi7280, 2019-05-17:
Sometimes I have a similar issue. I unplug the battery, plug it back in and that usually that fixes it. Very inconsistent. Holybro micro GPS, Airbot Omnibus F4 FC.

closed 2019-06-23 by stale[bot]

jasonslater1915, 2019-07-07:
im having a similiar issue.... i have the same gps module on mamba stack. wired up correcly.  module shows solid red light.  on jb's video it flashes blue stting its communicting with fc

aowi7280, 2019-07-08:
I reset defaults in the GPS and that fixed it. In BF Configurator, set the config to flying wing and turn off GPS auto config. Save and reboot. Go to the CLI and enter gpspassthrough, exit BF Configurator. Open u-blox u-center app, and connect. Then View/Configuration view. Click CFG (Configuration) and select the Revert to default configuration, then click send. Then I go back in to Rates and change Measurement Period to 100ms (10 Hz), then send. If it asks to go to the save page, do it and save the current configuration. Then go back to BF Configurator and change back to the Quad X or whatever you are flying.