betaflight/betaflight #8024: Bidirectional DSHOT: KAKUTEF7 & Tekko32 Dshot invalid pkts at 90-100% but osd shows esc RPM

avrahams, 2019-04-15:
I am trying to enable Bidirectional DSHOT on KAKUTEF7 with Tekko32 
I upgrade the esc firmware and used the default config snippet
but in the cli I see that almost all Dshot packages are invalid

I do see the ESC telemetry is the osd  (RPM, Temp, current etc) so I guess that the problem is not with my esc

dadoribeiro, 2019-04-15:
Some more info would be good, like BLHeli32 version (is it at defaults?), a diff all from the problem config. Did you try plugging the lipo and then USB afer (remove props), and then checking status in CLI, with and without motors spinning? The ESC telemetry you see has nothing to to with dshot bdir. Try defaults first before enabling options (like the ESC telemetry for OSD)

etracer65, 2019-04-16:
Even if 10% of the packets are valid you'll get RPM - just not very consistent or reliable data. So seeing RPM in the OSD doesn't mean "the problem is not your ESC". There is most definitely a problem with your ESC(s) and the communication if 90% of the telemetry packets are corrupted.

Please provide the details that were requested when you opened the issue.

etracer65, 2019-04-17:
What BLHeli32 firmware version do you have on your ESCs?

Dom4n, 2019-04-21:
I have this same setup, error rate is <1%.
My config is 4k/4k, dshot600. What are Yours?

hernandez87v, 2019-04-22:
Same setup here also, I'm using freestyle cli dump as suggested in wiki for freestyle. Also dshot600.