betaflight/betaflight #8026: CLracing F4 Mini - Softserial Smartport not working

SlipstreamFPV, 2019-04-16:
I've got a CLRacing F4 Mini and I've setup softserial.

Tested working via softserial: TBS Smartaudio, Runcam camera control

Tested, NOT working: Frsky smartport

Running the same port, on the same pin as the other devices doesn't work for the R9 Mini Rx smartport telemetry. Telemetry is obviously enabled and the link is working since the Rx voltage and RSSI transmit to the Taranis just not the smartport data.

The settings should be:

`feature SOFTSERIAL
feature TELEMETRY
tlm_halfduplex = on
tlm_inverted = off
resource SERIAL_TX 6 NONE
resource SERIAL_TX 11 C06
serial 30 32 115200 57600 0 115200
`

I've tried all combinations of tlm_halfduplex as well as tlm_inverted. ON, ON / ON, OFF / OFF, ON / OFF, OFF

I've got the same setup on multiple quads with other flight controllers and they all work fine.

I eleminated the Rx as a cause by swapping the R9 receiver from the CLracing F4 mini quad with another one from a quad with a DALF405. 
- The receiver from the CLracing F4 mini build worked fine in the DALF405 quad including smartport telemetry
- The receiver from the DALF405 quad didn't transmit smartport telemetry when hooked up to the CLracing F4 mini quad

I've tested BF 3.3 through to 4.0

Note: The R9 Rx is an original R9 Mini so there is no chance of uninverted smartport via hardware.

`# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Apr 11 2019 / 07:01:44 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name CLRACINGF4
manufacturer_id 

# resources
resource SERIAL_TX 3 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 11 C06
resource SERIAL_RX 4 NONE
resource SERIAL_RX 6 NONE

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 3 2048 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1800 2100 0 0
aux 1 13 2 1800 2100 0 0
aux 2 35 1 1400 1600 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_calibration = 26,26,-91
set rssi_channel = 8
set serialrx_provider = SBUS
set blackbox_device = NONE
set motor_pwm_protocol = DSHOT600
set bat_capacity = 520
set current_meter = ESC

# profile
profile 0

# rateprofile
rateprofile 0

set roll_srate = 78
set pitch_srate = 78

# end the command batch
batch end
`

SlipstreamFPV, 2019-04-17:
Turns out to be some issue with mapping softserial to Tx6 or Tx3 for smartport. Smartport via softserial works fine mapped to Rx6.

jflyper, 2019-04-17:
Tried to reproduce with Nucleo-F405RG flashed with CLRACINGF4 4.0.0 firmware.
R-XSR was used, SBUS connected to PA10 through an inverter, S.PORT connected to PC6 directly.
Gave following minimal CLI commands
```
resource SERIAL_TX 6 NONE
resource SERIAL_TX 11 C06
feature SOFTSERIAL
feature TELEMETRY
serial 0 64 115200 57600 0 115200
serial 30 32 115200 57600 0 115200
set serialrx_provider = SBUS
```

Result: The S.Port telemetry worked.
I noticed update frequency is some what slower than what it is in my memory, though.

There must be something that this setup is different from that exhibiting the issue.

mikeller, 2019-04-27:
@jflyper: Could this be another case affected by #8093?

jflyper, 2019-06-03:
Similar to the case of #8337, CLRACINGF4 target does not have valid timer entry for PC6, and a bug fixed by #8993 seems to have caused the invalid pin to work with 4.0.0.

closed 2019-06-03 by jflyper