betaflight/betaflight #8028: request to disable RSSI_ACD when using CRSF

TBSChin, 2019-04-16:
**Describe the bug**
The issue is i am getting 2-3 tickets a week from CRSF pilots who enable this setting when trying to get LQ/RSSI on OSD. it is not well described on betaflight, and it is not used at all when using CRSF.

**To Reproduce**
Put LQ on channel 12 of crossfire, select aux 8 for RSSI on the receiver tab of betaflight, enable RSSI on osd. then turn on RSSI_ADC, you will get incorrect information on your osd.

**Expected behavior**
LQ will not be correct.

**Flight controller configuration**
and FC

**Setup / Versions**
 -any version of betaflight

**Additional context**
due to the large amount of tickets i receive at TBS for this issue, i just wanted to see if it is possible to disable this setting when CRSF is chosen. it causes too many issues for a setting that isnt used with the protocol.

fujin, 2019-04-16:
spoke with @TBSChin regarding this, potentially affects DSMX as well (I assume certain DSMX radios / all DSMX radios do not have analog RSSI.)

etracer65, 2019-04-16:
It would make far more sense for this to be done in the configurator as a mutual lockout between RSSI_ADC and selecting a RSSI channel. Select one and the other disables. To make is more obvious the RSSI_ADC setting could be moved to the Receiver tab. Maybe improve the UI by having radio buttons to choose **between** analog RSSI and channel-based RSSI. @McGiverGim Thoughts?

mikeller, 2019-04-16:
Considering that CRSF is not a protocol that is exclusive to TBS (at least if TBS' repeated promises are to be believed), I do not think that anything should be done to restrict how users are configuring their gear just to gloss over obviously lacking documentation provided by one manufacturer.

@TBSChin: Can I suggest you instead provide better documentation to customers buying your gear, showing them clearly how to set up LQ / RSSI in OSD?

TBSChin, 2019-04-17:
@Mikeller rssi_adc wouldn’t have a use with crsf on other setups as rssi is not sent adc with crsf. So it doesn’t make sense tochave the option at all when using crsf.

And our documentation shows how to do it, I have short tutorials as well on the subject. It still doesn’t stop people who aren’t super familiar with betaflight and or crossfire fromcenabling this option. Then either complaining on fb that their lq gets low or submitting a ticket saying their lq either isn’t working or is getting extremely low. 

Even advanced pilots make this mistake often when coming from dsmx non rssi receivers.

mikeller, 2019-04-17:
@TBSChin:

> rssi_adc wouldn’t have a use with crsf on other setups as rssi is not sent adc with crsf. So it doesn’t make sense tochave the option at all when using crsf.

You really need to stop insinuating what other manufacturers will be doing when they are implementing the CRSF protocol. Betaflight do not have any interest in bowing to wishes from manufacturers if they behave like they are entitled to tell others what to do.

laz2wiringpi, 2019-04-17:
@mikeller 
i have been working on a native CRSF link stats 
if you could look at the code any thought's would be appreciated.
 i think @codecae has a crossfire i only have diversity nano to work with.

https://github.com/laz2wiringpi/betaflight/tree/crsf_link_stat

TBSChin, 2019-04-17:
@mikeller I’m not really sure where your hostility is coming from. #1 I’m not a manufacturer, I’m just a support agent asking for a feature due to the sheer amount of tickets I receive on the issue. #2 I’m not sure I ever “insinuated” anything. I was telling you the current case with crsf protocol how it is implemented. I do not know when crsf will be released to other manufacturers and have not heard anything about it in a while. I don’t see a release coming anytime soon. #3 I never told you to do anything, nor do I feel entitled to tell you anything. Simply a request with examples of the issue it is causing and asking what can be done about it. 

Currently, and for the foreseen future, there isn’t a reason to have the option when configuring crsf.

McGiverGim, 2019-04-17:
> It would make far more sense for this to be done in the configurator as a mutual lockout between RSSI_ADC and selecting a RSSI channel. Select one and the other disables. To make is more obvious the [...]

I think remember that @hydra talked about a modification that he had pending, of move the RX configuration to the receiver tab, if I'm not wrong.
The best will be make this complete refactor, move the RX, and all the RSSI fields to the receiver tab. In this way it has sense to block the RSSI analog field if the user selects the RSSI channel. Block fields of different tabs is not a good idea from the user view.

codecae, 2019-04-17:
@laz2wiringpi Unless something is changed, link statistics are not propagated or addressed from the RX to the FC.  The only place I've been able (historically) to obtain these frames are on the half-duplex RX transmissions between the TX Module and the TX and never on the RX side.  Has there been a change to the RX firmware that would start pushing the link statistics to the FC?  If so, is there any confirmation that *all available RX firmware* is now forwarding these frames to the FC?  If not, supporting a fragmented solution would probably just add to the issue queue.

codecae, 2019-04-17:
Maybe I'm missing something here but I have LQ on 12 and the RSSI OSD element is reacting fine to the data coming across....

codecae, 2019-04-17:
https://github.com/betaflight/betaflight/wiki/Configuring-Crossfire-LQ-for-use-as-RSSI-on-the-Betaflight-OSD

codecae, 2019-04-17:
See #8034

laz2wiringpi, 2019-04-17:
> @laz2wiringpi Unless something is changed, link statistics are not propagated or addressed from the RX to the FC. The only place I've been able (historically) to obtain these frames are on the [...]

yes i see link quality  and the other 10 values 

 Has there been a change to the RX firmware that would start pushing the link statistics to the FC?  
i dont know about past . it works.. matches exactly like radio values in telemetry..

laz2wiringpi, 2019-04-17:
@codecae 
" If so, is there any confirmation that all available RX firmware is now forwarding these frames to the FC?"
well sent me a picture of the data ..
this was from crossfire support 

I sniffed a frame for you. As I wrote earlier the link statistics frame is usually sent append an RC frame. The update rate of the link statistics frame is 10Hz. 

cant figure out how to post it ..
[image]

codecae, 2019-04-17:
I see that now... I hooked up my logic analyzer to the latest firmware and see that they send that frame @10hz.

Interesting.... I guess things happen when you take some time away :)

laz2wiringpi, 2019-04-18:
@codecae Could you take a look at my branch 
i do not  have a single antenna only density nano.
 and  im sort of stuck on the rssi dbm to percent  calculation .

codecae, 2019-04-18:
@laz2wiringpi  if you create a pull request, we can actually perform more formal peer reviews with inline commentary and code suggestions.

laz2wiringpi, 2019-04-18:
@codecae  yea still working it out. 
i have squashed it a few times you know how it is ... squash review .. Dam.. what was i thinking.... fix squash review ...  
So it ok to post a pr thats not complete .. ie work in progress ?
new to this community git stuff.

hydra, 2019-04-20:
@laz2wiringpi yes, and use appropriate labels if you have access to add them, like 'IN DEVELOPMENT', etc.  It also helps if you get run over by a bus too 💀 , since someone else can pick up where you left off!

hydra, 2019-04-20:
> I think remember that @hydra talked about a modification that he had pending, of move the RX configuration to the receiver tab, if I'm not wrong.

I did some work in the CF configurator that moved all receiver settings to one tab, but due to lack of time I have not ported this or re-done this for BF.  It's not difficult to do, just needs time...

mikeller, 2019-04-27:
@TBSChin:

> I’m not really sure where your hostility is coming from. #1 I’m not a manufacturer, I’m just a support agent asking for a feature due to the sheer amount of tickets I receive on the issue.

There is no hostility. I am just telling you that a single manufacturer (or their support agent or whatever) asking for an open source project to change its behaviour for something that seems to be a documentation issue with this manufacturer is out of line.

> #2 I’m not sure I ever “insinuated” anything. I was telling you the current case with crsf protocol how it is implemented. I do not know when crsf will be released to other manufacturers and have not heard anything about it in a while. I don’t see a release coming anytime soon.

If this is the case, this is a total departure from CRSF being opened up as a community driven protocol, like it was promised when we added support for it. We will probably have to consider removing support for CRSF from Betaflight, and replacing it with a proper open protocol. This will solve your problem.

#3 I never told you to do anything, nor do I feel entitled to tell you anything. Simply a request with examples of the issue it is causing and asking what can be done about it.

Currently, and for the foreseen future, there isn’t a reason to have the option when configuring crsf.

Ok, if this is TBS's answer to when CRSF will be opened up as originally promised, I will open a pull request to have it removed from Betaflight.

hydra, 2019-04-27:
> Ok, if this is TBS's answer to when CRSF will be opened up as originally promised, I will open a pull request to have it removed from Betaflight.

I'm confused; remove what? CRSF entirely? Or am I mis-reading here?

mikeller, 2019-04-28:
@hydra: Well, if it is as @TBSChin is stating ('I don’t see a release coming anytime soon.'), then this clearly runs counter to what TBS was promising when CRSF support was initially added to Betaflight, which was that CRSF would be opened up and made available as a protocol to other manufacturers. As such, and considering TBS does not seem to be happy how CRSF is implemented in Betaflight, I think we will have to reconsider if we want to keep supporting it in Betaflight, or if we should look into replacing it with a non-proprietary alternative.

TBSChin, 2019-04-29:
@mikeller once again, I am a support agent, I do not control what tbs does, nor am I privy to any upcoming releases. This isn’t tbs’s answer to anything. I know as much as you do. I just solve problems. I do not understand your hostility to me asking if there is a way to fix this issue. 

What I have learned from this whole interaction is that because I hold a job somewhere. I am not able to suggest anything as a community member and person without a betaflight dev coming at me for what the company I work for does. 

I will no longer be posting here and I’d rather just move on and have you ignore any suggestion I have had.

I personally am kind of taken aback by the way you have responded to a simple inquiry. I don’t expect change, I don’t expect anything, and I see that I have nothing to contribute to this conversation. To the point I will no longer attempt to contribute.

tbs-trappy, 2019-04-29:
A few statements have been interpreted to have been made on behalf of TBS, which is incorrect. For the record

1) TBS is not unhappy with how CRSF is implemented in Betaflight, and never has been, or has ever said anything to that regard.
2) CRSF is not a proprietary protocol. Anyone who has asked us about implementation details has received them, and has received support to the best of our abilities. 
3) The plan to release a CRSF library into open source is still under way, we have several hundred developer hours into this effort and still counting. Things take time, there are unforseen delays, new feature requests and shifts in priorities and I hope that as developers you guys can appreciate this.

The OP has absolutely no relevance to how CRSF is implemented in betaflight. It was a trend that was spotted by our CS of people struggling with this setting, and he was simply trying to help. Thanks everyone for your time.

mikeller, 2019-04-29:
@tbs-trappy:

> CRSF is not a proprietary protocol. Anyone who has asked us about implementation details has received them, and has received support to the best of our abilities.
> The plan to release a CRSF library into open source is still under way, we have several hundred developer hours into this effort and still counting. Things take time, there are unforseen delays, new feature requests and shifts in priorities and I hope that as developers you guys can appreciate this.

Glad to hear that.

In this case I guess I should make another effort on dusting off #6917? I'd let this go to sleep after not hearing anything about the efforts to open this up for a long time.

I assume this request can then be closed, as it does not make sense to enshrine details of one particular hardware implementation into the behaviour of firmware or configurator?

tbs-trappy, 2019-04-29:
I don't think work is needed on #6917 - we are planning to release a library into open source which can then be merged into betaflight. this is a separate issue, I'm available to discuss the further steps but we'll probably still need a few more weeks to get everything ready.

I do agree that hardware implementation and configurator behavior should remain separated. Mike Chin simply made a suggestion. The decision to modify or not is - and has always been - yours. We are fine either way, and will work on our end to mitigate any issues arising from this.

mikeller, 2019-04-29:
@tbs-trappy: All good, I'll hold off on #6917 then.

I understood Mike's original post as a request (which is what he titled it), which I thought was out of line.

tbs-trappy, 2019-04-29:
certainly a misunderstanding. just trying to provide feedback.

TBSChin, 2019-04-29:
@mikeller allow me to apoligize then, as this was what seems a major misunderstanding. I was told by Aj Christensen to post this here when I messaged him asking him if there was a way to make it easier. He said to post it here, in this format. I have never used the betaflight github, this was my first post.

I wasn’t trying to step on toes, or tell people what to do, I was just sharing my experience with something I have seen pop up. 

Like I said in my previous comment, I just work support and fix things. I’m not high enough up the ladder to know what’s coming and when. I understand your frustration, under the thought that what I am saying represents tbs, but i don’t.

I’d love to start on a new page with you and squash whatever has happened here, I don’t mean to annoy you or cause frustration on your end. 

Sorry for the miscommunication.

Can you and I pets

mikeller, 2019-04-29:
@TBSChin: All good now. Unfortunately it is quite common for manufacturers to try to get changes into Betaflight that only help them to the detriment of users and the volunteers doing work for Betaflight, so my apologies for reacting harshly to what I perceived as one more attempt at this.

Work is being done on #8042, which aims to give native RSSI / LQ capabilities to CRSF, which should hopefully fix this support issue - and increase the value that CRSF users get beyond that.

TBSChin, 2019-04-29:
@mikeller I sometimes forget that people see my opinions or suggestions as a suggestion from tbs itself. I’ll keep that in mind when I’m interacting with companies/developers. 

I’m glad we smoothed this out.

closed 2022-06-01 by haslinghuis