betaflight/betaflight #8030: AKK - RDQ VTX Patch for Betaflight 4.0

wurstyboy, 2019-04-16:
Currently I did not find any "AKK - RDQ VTX Patch" version of BF4?
Is this scheduled already? Or will BF4 without support for the buggy smartaudio protokol of AKK and RDQ?

mikeller, 2019-04-16:
@wurstyboy: Yes, this is planned. @wind0r is working on this.

wind0r, 2019-04-27:
It started building: https://ci.betaflight.tech/job/Betaflight%20(AKK%20-%20RDQ%20VTX%20Patch)/

ethomas997, 2019-04-27:
@mikeller @wind0r Right now the 'akk_vtx_fix' version of 'saSendFrame()' in "io/vtx_smartaudio.c" looks like this:
```
static void saSendFrame(uint8_t *buf, int len)
{
    if (!IS_RC_MODE_ACTIVE(BOXVTXCONTROLDISABLE)) {
        switch (smartAudioSerialPort->identifier) {
            case SERIAL_PORT_SOFTSERIAL1:
            case SERIAL_PORT_SOFTSERIAL2:
                break;
            default:
                serialWrite(smartAudioSerialPort, 0x00); // Generate 1st start bit
                break;
        }

        for (int i = 0 ; i < len ; i++) {
            serialWrite(smartAudioSerialPort, buf[i]);
        }

        saStat.pktsent++;
    } else {
        sa_outstanding = SA_CMD_NONE;
    }

    serialWrite(smartAudioSerialPort, 0x00);

    sa_lastTransmissionMs = millis();
}
```

Shouldn't the "serialWrite(smartAudioSerialPort, 0x00);" line be added right after the 'for' loop, so it's only invoked when BOXVTXCONTROLDISABLE=off, like this?:
```
static void saSendFrame(uint8_t *buf, int len)
{
    if (!IS_RC_MODE_ACTIVE(BOXVTXCONTROLDISABLE)) {
        switch (smartAudioSerialPort->identifier) {
            case SERIAL_PORT_SOFTSERIAL1:
            case SERIAL_PORT_SOFTSERIAL2:
                break;
            default:
                serialWrite(smartAudioSerialPort, 0x00); // Generate 1st start bit
                break;
        }

        for (int i = 0 ; i < len ; i++) {
            serialWrite(smartAudioSerialPort, buf[i]);
        }

        serialWrite(smartAudioSerialPort, 0x00);

        saStat.pktsent++;
    } else {
        sa_outstanding = SA_CMD_NONE;
    }

    sa_lastTransmissionMs = millis();
}
```

mikeller, 2019-04-27:
@wurstyboy, @RaffaeleNachos: All available now:

[image]

@ethomas997: Yes, you are right. Looks like a cherry-pick went wrong without throwing a merge conflict. Do you want to open a pull request against the `akk_vtx_fix` to fix this?

ethomas997, 2019-04-29:
@mikeller Sure, posted https://github.com/betaflight/betaflight/pull/8124

RipperDrone, 2019-05-29:
Sry to ask again: What's the BF4.1 strategy to mitigate all different SA protocols and botched up AKK/RDQ versions? Will confusion @ TBS protocols SA lite, 1.0, 2.0, 2.1 lead to a common CONFIGURABLE solution which encompasses AKK/RDQ "need pull down bit at end of frame"? I remember this has been communicated as the final targeted solution, however since seeing the VTX table implementation I'm no longer sure... 🤔

closed 2019-07-05 by stale[bot]

dentex, 2019-08-04:
Hi. 
For these type of VTXs, do we still need the pached version, of can BF 4.x handle them? Thanks

mikeller, 2019-08-05:
@dentex: Betaflight 4.0 was going to support AKK VTX. But unfortunately TBS released a VTX (TBS Unify Pro with SmartAudio 2.1) that will not work with the version of the protocol that supports AKK VTX, so we are back to square one when it comes to making this work for all VTX.

dentex, 2019-08-05:
@mikeller Thanks. I was asking in regards of some obscure VTX I have (so no TBS Unify...), that probably it's an AKK knock-off:
https://www.unmannedtechshop.co.uk/product/chaosfpv-vtx-t-video-transmitter-25-800mw-7-24v-betaflight-telemetry/?attribute_pa_connector-style=sma-pigtail
Sorry for the product link, it's just to show they advice to use the AKK patch.
I did not actually, until v4.0.3. Yesterday I updated to a version built from your `master` brach and find out I cannot change channel, power levels or anything via OSD. The phisical button worked.
I was using this vtxtable:

'''# vtxtable
vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A CUSTOM  5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B CUSTOM  5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E CUSTOM  5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F CUSTOM  5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R CUSTOM  5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 4
vtxtable powervalues  25 200 500 800
vtxtable powerlabels 25 200 500 800'''

Then I didn't have a chance to investigate further.

mikeller, 2019-08-06:
@dentex: The problem is there are a number of VTX out there that implement more or less correct versions of the SmartAudio protocol - the protocol spec is not 100% clear to start with. Some are broken with the code in `master` and work with the code in `akk_vtx_fix`, for others it's the other way round, yet others do not work completely with either. (The same goes for VTX implementing the IRC Tramp protocol.)
So I suspect as long as manufacturers throw cheaply designed VTX onto the market that can not be firmware upgraded this problem will not be solvable.

dentex, 2019-08-06:
@mikeller yep I imagined the cheap VTX might be the problem :)
I'll check building from an older version and testing on the actual VTX if something changed recently in `master`. Then I'll eventually look into the `akk_vtx_fix` branch. For the moment thanks again for your attention. Bye!