betaflight/betaflight #8031: RPM Filtering for Kakute F7

oldrootbeer, 2019-04-16:
Hi. Tried the default and Kakute F4 snippets for my Kakute F7.
Invalid packets close to 100%.
Can you help? Dynamic resonance filtering looks like a sound way to go.

oldrootbeer, 2019-04-16:
# diff all

# version
# Betaflight / KAKUTEF7 (KTF7) 4.0.0 Apr 11 2019 / 07:20:40 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name KAKUTEF7
manufacturer_id 
mcu_id 002700203436510b34333434
signature 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY
feature ESC_SENSOR

# beeper

# beacon
beacon RX_LOST

# map

# serial
serial 0 64 115200 57600 0 115200
serial 1 2048 115200 57600 0 115200
serial 2 2 115200 57600 0 115200
serial 5 0 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 1700 0 0
aux 1 1 2 900 1200 0 0
aux 2 46 1 1675 2100 0 0
aux 3 35 3 1675 2100 0 0
aux 4 47 2 1300 1700 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 0
set gyro_lowpass2_type = PT1
set gyro_lowpass2_hz = 150
set dyn_notch_width_percent = 0
set dyn_notch_q = 250
set dyn_lpf_gyro_min_hz = 0
set acc_calibration = 51,30,34
set mag_hardware = NONE
set rssi_channel = 14
set serialrx_provider = CRSF
set dshot_idle_value = 350
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1500
set vbat_scale = 108
set ibata_scale = 288
set gps_provider = UBLOX
set pid_process_denom = 1
set osd_warn_batt_not_full = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_cap_alarm = 1000
set osd_vbat_pos = 2066
set osd_rssi_pos = 25
set osd_link_quality_pos = 2074
set osd_crosshairs_pos = 2253
set osd_ah_pos = 2126
set osd_mah_drawn_pos = 2059
set osd_gps_sats_pos = 7
set osd_home_dir_pos = 2094
set osd_home_dist_pos = 2082
set osd_altitude_pos = 2049
set osd_stat_rtc_date_time = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_battery = ON
set osd_stat_max_alt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set osd_stat_min_link_quality = ON
set debug_mode = RPM_FILTER
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_freq = 5658
set vcd_video_system = PAL

# profile
profile 0

set dterm_lowpass2_hz = 0
set d_min_roll = 27
set d_min_pitch = 29
set d_min_boost_gain = 35

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_mid = 29
set thr_expo = 60

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

oldrootbeer, 2019-04-16:
[image]

oldrootbeer, 2019-04-16:
blheli 32.6. Motors 1 & 4 are reversed.

etracer65, 2019-04-17:
Please read the instructions. https://github.com/betaflight/betaflight/wiki/Bidirectional-DSHOT-and-RPM-Filter

You must update your ESC's with the beta BLHeli32 firmware.

oldrootbeer, 2019-04-17:
Whoops! 32.61 now loaded. Thank you. A little more attention to the fine print next time eh? :-D

closed 2019-04-17 by oldrootbeer