betaflight/betaflight #8032: Reciever entering binding on boot

JoBoWi, 2019-04-16:
Hey guys,

my receiver is a R617XL - DSM2 / x 6CH PPM receiver.
Everytime I plug my battery in, it enters bind mode automatically.
When I bind it, it shows up in my controls and everything works fine.
Until the next power-cycle. Then it boots into bind again.
I use Betaflight 4.0.0 on a HAKRC BF3.2 Omnibus F4 if that helps.
I tried using the spektrum_sat_bind = 0 command to see if it helps.
However that just spits out an error. Same for set spectrum_sat_bind_autorst = 1.
I ve tried just hooking the receiver to 5V and it works fine whitout entering bind.

What am I doing wrong here?

watageek, 2019-04-16:
@mikeller
Do you think the signal wire is grounding  while it power up because the pin is pulled low on boot? Putting the receiver in bind mode.

etracer65, 2019-04-17:
The parameters are `spektrum_sat_bind` and `spektrum_sat_bind_autoreset`.

AndersHoglund, 2019-04-17:
This is a PPM Rx, i.e. analog serial PWM, right?  Not a digital UART connected SerialRX Rx. 
Then the `spektrum_sat_bind*` parameters are NOT applicable. 

Binding this Rx is done by shortening signal to gnd, so something is pulling it low during boot.  

How and where is this Rx connected to the FCs?  Please also supply BF config details by added the output from the diff cli command, as you where asked to when opening an issue.

Note:; I do not have any such Rx, so I can not verify nor help testing this issue.  Can only speculate with the little information given.

closed 2019-05-24 by stale[bot]