betaflight/betaflight #8036: GPS Rescue FS crash detection

Jetrell, 2019-04-18:
From my tests. It appears that GPS rescue does not have a safety motor shut-off.  If a crash was to occur while Auto Navigating to the takeoff location, while under an RX Failsafe condition.

If a crash was to occur. It appears that the motors would keep spinning.
I know this is an edge case. But it is a possibly, if the TX or RX failed and your were unable to take control again, or able to get to the quad in a hurry, to shut it down. 
It could burn-out a motor or ESC.

Is there any contingency for this condition that I have not noticed? 

If not....Would it be possible to have the flight controller detect no reactions to its control inputs, because its on the ground. Plus detecting a high "G" impact right before this. 
Then it would then shut the motors off. 
And if the GPS is still functioning. It could also supplement this with GPS Speed and Altitude data, to farther decrease the likely hood of a false motor shutoff condition.

McGiverGim, 2019-04-18:
In theory, the sanity checks must do it, but maybe is slower than detecting a crash, it disarms the quad if something does not work as expected. I've not tested it: `set gps_rescue_sanity_checks = RESCUE_SANITY_ON`

Jetrell, 2019-04-18:
@McGiverGim  That's nice to know `gps_rescue_sanity_checks = ON` is supposed to disarm if a FS crash occurs. I just didn't know if it was written into the code, to take into account that condition?

I understand that it looks for the GPS data not fitting with the expected outcome. 
e.g. distance to home getting > and not <.  And if Altitude is rising, not sinking. 
But as for a crash on the way?
The Wiki says nothing of it, that's all.

Thanks for your reply

McGiverGim, 2019-04-18:
What I wanted to say is that if you crash, the outcome will not be reached and it will disarm, but maybe it takes some second.

closed 2019-04-18 by Jetrell