betaflight/betaflight #8041: GPS rescue climbs to 125m, while initial_alt is set to 35m (BF4.1)

krzysztofmatula, 2019-04-18:
DVR: https://www.dropbox.com/s/md4h5mde3bdiagk/PICT0092-RESC.AVI?dl=0
LOG: https://www.dropbox.com/s/bxf6hqymsb7x8g8/LOG00079-RESC.BFL?dl=0

It occurred twice: at 3:00 and at 3:44.
After engaging GPS Rescue by switch the quad first climbed to 125m, then started cruising towards home. I'd expect it will climb to 35m, as I have `gps_rescue_initial_alt = 35`.

I was testing this configuration around 4.0-RC2 and it worked fine.
Now I'm at the latest 4.1 master.

My complete gps rescue config:
```
set gps_rescue_angle = 32
set gps_rescue_initial_alt = 35
set gps_rescue_descent_dist = 60
set gps_rescue_landing_alt = 5
set gps_rescue_landing_dist = 10
set gps_rescue_ground_speed = 1200
set gps_rescue_throttle_p = 150
set gps_rescue_throttle_i = 20
set gps_rescue_throttle_d = 50
set gps_rescue_velocity_p = 80
set gps_rescue_velocity_i = 20
set gps_rescue_velocity_d = 15
set gps_rescue_yaw_p = 40
set gps_rescue_throttle_min = 1200
set gps_rescue_throttle_max = 1600
set gps_rescue_throttle_hover = 1370
set gps_rescue_sanity_checks = RESCUE_SANITY_ON
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_use_mag = OFF
```

Full diff:
```
# diff

# version
# Betaflight / MATEKF722SE (MF7S) 4.1.0 Apr 15 2019 / 20:42:00 (2654bf07b) MSP API: 1.42

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id 

# name
name KM

# resources
resource PWM 1 NONE
resource PWM 2 NONE
resource SERIAL_TX 2 NONE
resource SERIAL_TX 11 A02
resource SERIAL_RX 4 NONE
resource CAMERA_CONTROL 1 A01

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature GPS
feature ESC_SENSOR

# beeper

# beacon

# map

# serial
serial 2 2 115200 115200 0 115200
serial 3 16384 115200 57600 0 115200
serial 5 1024 115200 115200 0 115200
serial 30 16 115200 57600 57600 115200

# led

# color

# mode_color

# aux
aux 0 0 5 1550 2100 0 0
aux 1 2 1 900 1200 0 0
aux 2 46 0 1550 2100 0 0
aux 3 13 1 1550 2100 0 0
aux 4 15 2 1700 2100 0 0

# adjrange

# rxrange
rxrange 0 1019 2019
rxrange 1 1019 2019
rxrange 2 1019 2019

# vtx

# rxfail

# display_name

# master
set dyn_lpf_gyro_max_hz = 620
set acc_calibration = 34,-27,71
set mag_hardware = NONE
set rssi_channel = 9
set blackbox_p_ratio = 64
set min_throttle = 1050
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT1200
set ibata_scale = 204
set small_angle = 180
set gps_provider = UBLOX
set gps_auto_baud = ON
set gps_rescue_initial_alt = 35
set gps_rescue_descent_dist = 60
set gps_rescue_ground_speed = 1200
set gps_rescue_throttle_hover = 1370
set gps_rescue_min_sats = 5
set gps_rescue_min_dth = 50
set gps_rescue_allow_arming_without_fix = ON
set gps_rescue_use_mag = OFF
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1400
set osd_tim1 = 2563
set osd_vbat_pos = 2402
set osd_rssi_pos = 2089
set osd_tim_1_pos = 2454
set osd_tim_2_pos = 406
set osd_flymode_pos = 2448
set osd_anti_gravity_pos = 2422
set osd_throttle_pos = 24
set osd_crosshairs_pos = 2253
set osd_current_pos = 2102
set osd_mah_drawn_pos = 2441
set osd_gps_speed_pos = 2241
set osd_gps_sats_pos = 2080
set osd_home_dir_pos = 2094
set osd_home_dist_pos = 2097
set osd_altitude_pos = 2262
set osd_warnings_pos = 14697
set osd_avg_cell_voltage_pos = 2434
set osd_esc_tmp_pos = 122
set osd_esc_rpm_pos = 87
set osd_core_temp_pos = 154
set osd_stat_max_dist = ON
set osd_stat_min_batt = OFF
set osd_stat_max_fft = ON
set osd_stat_total_flights = ON
set osd_stat_total_time = ON
set osd_stat_total_dist = ON
set debug_mode = DYN_LPF
set vcd_video_system = NTSC
set stats = ON
set stats_total_flights = 36
set stats_total_time_s = 7944
set stats_total_dist_m = 85001
set name = KM

# profile
profile 0

set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 200
set feedforward_transition = 30
set iterm_relax_type = GYRO
set p_pitch = 55
set i_pitch = 85
set p_roll = 52
set i_roll = 80
set d_roll = 30
set f_yaw = 35
set d_min_roll = 25
set d_min_pitch = 28
set d_min_boost_gain = 30
set d_min_advance = 50

# rateprofile
rateprofile 0

set thr_mid = 35
set thr_expo = 30
set roll_rc_rate = 150
set pitch_rc_rate = 150
set yaw_rc_rate = 150

# end the command batch
batch end
```

IllusionFpv, 2019-04-18:
https://github.com/betaflight/betaflight/blob/55a5d70cc84004a8f5946242ed754d2925e228de/src/main/flight/gps_rescue.c#L543

gps_rescue_initial_alt = 35 will be substituded by max altitude reached in flight + 15mt

krzysztofmatula, 2019-04-18:
OK, understand. So it seems to work well then.
Apart from the very last attempt to engage rescue that disarmed mid-flight.
I know some say it is a "feature", but I can't agree with this as I just did it another time :(

closed 2019-04-18 by krzysztofmatula

etracer65, 2019-04-18:
At 2:20 in the log you climbed to over 100m. Looks like it worked the way it should.

IllusionFpv, 2019-04-18:
@krzysztofmatula in the last attempt you engaged gps rescue too close from home, gps rescue will works after `gps_rescue_min_dth = 50`, Otherwise quad will disarm in mid-flight

krzysztofmatula, 2019-04-18:
Yes, I know it...
I will always vote for ignoring the switch in such case instead of disarming.
I try to remember to check distance while testing, but I always forget at some point :(