betaflight/betaflight #8045: Use iNav fixes to SBUS handling

krzysztofmatula, 2019-04-19:
A year ago some work has been done in iNav (mainly by @digitalentity) to improve SBUS handling code. Consider taking the fixes to BF.

One known issue is that there are RX devices (like eLeReS RX) that are unable to meet strict timeout requirements expected by the current code. For those RX-es one need to tweak the following define:
```
#define SBUS_TIME_NEEDED_PER_FRAME 3000
```

Since iNav improvements this is no longer needed (the define no longer exists).

etracer65, 2019-04-19:
iNav still has this timeout (and the default is still 3ms). They just made it configurable with the parameter `sbus_sync_interval`.

krzysztofmatula, 2019-04-19:
I think it is not the same timeout (`SBUS_TIME_NEEDED_PER_FRAME` vs `SBUS_DEFAULT_INTERFRAME_DELAY_US`).
For sure no tweaks are needed for eLeRes to work on iNav now.

etracer65, 2019-04-19:
The SBUS spec is 100k baud, 8E2 async serial. The timeout in Betaflight is actually `SBUS_TIME_NEEDED_PER_FRAME` + 500us so 3500us. For a frame to timeout the receiver would have to be transmitting at < ~86k baud - way outside the spec! If there are receivers out there that are this poorly implemented I don't see why we should deviate from the specifications just to provide a workaround for them.

And taking a quick look at our SBUS implementation I don't see any logic to enforce any minimum delay between frames.

krzysztofmatula, 2019-04-19:
For this particular RX it maintains 100kbps, but can't guarantee no gaps between bytes. It also maintains Futaba 14ms frame interval. In practice it transmits full message (22 bytes) within 3-6ms.

It's not against any specs according to the RX devs.

closed 2019-05-26 by stale[bot]