betaflight/betaflight #8047: LED strip cannot be configured for SPRACINGF3

joefernwright, 2019-04-19:
It is not possible to save config for LED strip using spracingf3 hardware. Save button just does not save any settings. Had to roll back to the previous firmware version where led strip config works fine.

jflyper, 2019-04-19:
Please follow the instruction in the preformatted text given when you first opened this issue.

You are not even saying which firmware version you have the issue with.

closed 2019-04-19 by jflyper

jflyper, 2019-04-19:
If you are referring to 4.0 as non working version, LED strip as status indication was dropped from F3.

https://github.com/betaflight/betaflight/releases/tag/4.0.0
> support for status display through LED strips was removed from F3 based flight controllers, in order to be able to reclaim some flash space. In its place, LED Strip Profiles (except for the 'Status' profile which is not supported on F3) can be used to set LED strips to fixed colours.

fremebondo, 2019-09-28:
Same here, and I cannot find nothing related to Profiles
[image]