betaflight/betaflight #8048: Move filtering under named profiles

mavericm1, 2019-04-19:
**Is your feature request related to a problem? Please describe.**
Betaflight defaults cannot both provide safe flight and aggressive performance.  Dev's are always trying to give the best possible settings for flight by default which is not possible with the wide variety of well build and horribly built quads out there

**Describe the solution you'd like**
I'd like to move all dyn and gyro filtering under the pid profiles with the understanding that gyro and dyn filters cannot change mid flight.   I'd also like a name variable added to profiles so that a profile can be named.   This will allow multiple presets in the default flash of betaflight a safe default profile setup and a performance based profile setup.   By easily selecting a profile to use the user can achieve better flight easily.  This allows to focus on more aggressive default settings that are suitable for quads that are built well and safe settings for those that need them.

**Describe alternatives you've considered**
This is already partially being done by config snippets in the wiki but i think this a better way forward for profiding defaults that can be safe and ones that are forcused on flight performance

**Additional context**
we know safe defaults that work for most every quad out there this will allow us to stop focusing on trying to get the best settings for everything out there and just allow dev's and betaflight community to move forward with focusing on settings for performance 

Please note that feature requests are not 'fire and forget'. It is a lot more likely that the feature you would like to have will be implemented if you keep watching your feature request, and provide more details to developers looking into implementing your feature, and help them with testing.

hsuq510, 2019-05-17:
any plan to implement this ?

closed 2019-06-23 by stale[bot]

babipsylon, 2020-12-11:
Yes please! Multiple filter profiles would be great and so useful!!