betaflight/betaflight #8050: modes

fragged8, 2019-04-19:
**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

fragged8, 2019-04-19:
modes screen totally blank. no way to configure switches
[image]

jflyper, 2019-04-20:
I suspect your configuration is corrupt. Goto CLI tab and enter "defaults" and immediately examine if the issue persists. If not, start configuring from scratch, NOT by simply pasting old configuration from previous releases.

etracer65, 2019-04-20:
You’re running an old 4.0 dev version from December. Update to the official 4.0 release.

closed 2019-04-30 by etracer65