betaflight/betaflight #8051: Warning: no gyro/Acc

fragged8, 2019-04-19:
**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

fragged8, 2019-04-19:
Mobius 7 . Config tab, No Gyro/ Acc 
[image]

tenten8401, 2019-04-20:
Mine light up yellow in the top too, and my quad flies fine. That message might just be a bug in the configurator.

[image]

jflyper, 2019-04-20:
I suspect your configuration is corrupt. Goto CLI tab and enter "defaults" and immediately examine if the issue persists. If not, start configuring from scratch, NOT by simply pasting old configuration from previous releases.

etracer65, 2019-04-20:
You’re running an old 4.0 dev version from December. Update to the official 4.0 release.

fragged8, 2019-04-23:
thanks for that, I downloaded from the info box that appeared in betaflight .. will look into it

etracer65, 2019-04-23:
I'm talking about the firmware version on your flight controller - not the configurator version. There is no "info box" that prompts you to upgrade the firmware. You are running a nightly build from December - there is no guarantee that nightly builds will continue to be supported. Upgrade to the official Betaflight 4.0 release which was released 11 April 2019. The version you're running is 4 months old.

[image]

fragged8, 2019-04-23:
ok, thank you I will look at that this evening ..

closed 2019-05-30 by stale[bot]