betaflight/betaflight #8052: osd tab blank

fragged8, 2019-04-19:
**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

fragged8, 2019-04-19:
OSD tab opens blank page. 
[image]

jflyper, 2019-04-20:
I suspect your configuration is corrupt. Goto CLI tab and enter "defaults" and immediately examine if the issue persists. If not, start configuring from scratch, NOT by simply pasting old configuration from previous releases.

McGiverGim, 2019-04-20:
No, your quad has a no official version of Betaflight 4.0. Ask your manufacturer for a tool to configure it out upgrade the firmware to the official 4.0 version.

jflyper, 2019-04-20:
@McGiverGim I think it does.
[image]

chung212, 2019-04-20:
I have same Issue after update to latest betaflight build
before #1565 no this Issue

jflyper, 2019-04-20:
@chung212 CI build #1565 only updated EEPROM version information, and if you have a problem starting with this build, you are having corrupt configuration. Start with fresh config on fresh flash, up-to-date configurator and skill to read and execute standard procedures for trying development build.

chung212, 2019-04-20:
NO the problem was from #1566
I already update to latest configurator and development build.
so I need to back to #1565

chung212, 2019-04-20:
problem solve
My issue is from configurator
when I use configurator build 160 no problem
but change to build 161 or above ,have  problem

chung212, 2019-04-20:
this mean when use latest configurator ,betaflight only use to build 1565
when use configurator build 160,It can use latest betaflight build

etracer65, 2019-04-20:
[image]

It's known that old 4.0 dev builds have problems with the current Configurator. Update to the current 4.0 release - I don't mean a dev build, I mean official release.

closed 2019-04-30 by etracer65