betaflight/betaflight #8053: f722se adc and voltage scales changing on disarm

NITROSPORTMAN, 2019-04-20:
matek f722se upon disarm voltage and current sensor values change

put an f722se on 4.0 and arm and disarm

video
https://www.youtube.com/watch?v=bFI3rs8zv4s&feature=youtu.be&fbclid=IwAR3iOcPzxQKo2cYwDgK9akHuWIN5A0j09OlPNsLQIari_cG17l5gvk5UoRs

values should not change

# dump

# version
# Betaflight / MATEKF722SE (MF7S) 4.0.0 Apr 11 2019 / 07:26:45 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id 

# name
name -

# resources
resource BEEPER 1 C13
resource MOTOR 1 B04
resource MOTOR 2 B05
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 A15
resource MOTOR 6 B03
resource MOTOR 7 B06
resource MOTOR 8 B07
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 A03
resource PWM 1 A02
resource PWM 2 A01
resource PWM 3 A00
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 A08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 NONE
resource SERIAL_RX 6 C07
resource SERIAL_RX 7 NONE
resource SERIAL_RX 8 NONE
[... 889 lines cut ...]
rateprofile 0

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 50
set tpa_breakpoint = 1500
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# end the command batch
batch end

Docteh, 2019-04-21:
What happens if you simply restart the flight controller? Also are these the same numbers each time? I'd try cutting down on the items showing in your "Post flight statistics", and report back the result, pass or fail. If that is the right direction, maybe paste in a picture of the "post flight statistics" screen into this issue?

77 65 88 spells  MAX

21317 could be "SE"
8259 could be " C"

```
osd/osd.c:    if (osdStatGetState(OSD_STAT_MAX_ESC_TEMP)) {
osd/osd.c:        osdDisplayStatisticLabel(top++, "MAX ESC TEMP", buff);
osd/osd.c:    if (osdStatGetState(OSD_STAT_MAX_ESC_RPM)) {
osd/osd.c:        osdDisplayStatisticLabel(top++, "MAX ESC RPM", buff);
```

Remember: I could be barking up the wrong tree.  😎

NITROSPORTMAN, 2019-04-21:
Restarting after changes nothing it's almost if there is a back end changing stuff

Docteh, 2019-04-21:
Any thoughts on cutting down the items to show in the post flight statistics?

Docteh, 2019-04-21:
Looks like the Final Statistics page is writing past the end of the screen. It probably works fine in 3.5.7 because something else is getting overwritten, instead of configuration parameters. 

https://github.com/betaflight/betaflight/blob/9ad2cc84452b8dcfa385bc5d8ea3cf9ab86ad046/src/main/osd/osd.c#L443-L448

osdDisplayStatisticLabel doesn't limit y

https://github.com/betaflight/betaflight/blob/72383f3064d497959fc542aa78b0b4724225626c/src/main/drivers/display.c#L79-L84

displayWrite doesn't. 

https://github.com/betaflight/betaflight/blob/aed1e51cf6588630ff847ba460ca4d0a6397a933/src/main/io/displayport_max7456.c#L103-L109

Last chance is maybe max7456Write will do it. 

https://github.com/betaflight/betaflight/blob/eba58566545038d0f5f55b1a0ce203065c69a74a/src/main/drivers/max7456.c#L577-L584

laz2wiringpi, 2019-04-21:
@Docteh nice find are you working  pr for this?
any maintainers please make as bug? looks like its in all versions?

NITROSPORTMAN, 2019-04-21:
Will change up post flight statistics and report back.

PR #8063 (2019-04-28): prevent max7456 screen_buffer from overflow
Hi, this should fix #8053 

Since it is preventing an overflow of a buffer, I figure it should be closest to where the buffer is written to. 

However my particular drone is short an item or two to properly test...
[image]
closed 2019-04-28 by mikeller