betaflight/betaflight #8054: There's no OSD tab in 10.5.1

Ion-FPV, 2019-04-20:
**Important: This is not the right place to get help with configuration issues. Your issue will be closed without further comment.**

I opened up betaflight 10.5.1 to try to put a custom logo on my new Talon F7, when I plugged it in to see that there was no OSD tab. I tried closing and reopening the software, but that didn't fix anything. I'm just confused because it was there yesterday but not today. 

[image]

Ion-FPV, 2019-04-20:
Should I re-download the new configurator?

McGiverGim, 2019-04-20:
No, your quad has a no official version of Betaflight 4.0. Ask your manufacturer for a tool to configure it out upgrade the firmware to the official 4.0 version.

jflyper, 2019-04-20:
@McGiverGim Talon F7 is using CLRACINGF7 target.

@Ion-FPV Which version of firmware are you using?

etracer65, 2019-04-20:
Could just be as simple as the OSD feature not enabled on the Configuration tab. But without a `diff` of settings as required when opening an issue it's impossible to help.

Ion-FPV, 2019-04-20:
@jflyper I'm running 4.0

etracer65, 2019-04-20:
Provide the `diff` as required or we cannot help any further.

Ion-FPV, 2019-04-20:
Sorry @etracer65, you were right. I forgot to turn on the osd switch. I just assumed is was on by default. Thanks for all of your responses.

closed 2019-04-20 by Ion-FPV