betaflight/betaflight #8055: Betaflight 4.0 NO Gyro with Matek System F722-SE

Oko001, 2019-04-20:
The FC works well with previous Betaflight versions. After install the 4.0 FW, the accelerometer is working, and calibrateing well. But when I set up the UART1 for my GPS, OR UART3 for TBS smart audio, the accelerometer and the gyro is disappear. Reset, uart switch off not help, just reflash for get accelometer back.
  I had the same issue with 4.0 release candidates too.

I use the Matek System F722-SE with the MPU6000, 8KHz/8Khz with GPS, and AKK smart audio vtx, XM+ receiver, Racerstar PG50x4 50A 2-6S Blheli_32 Proshot 4 In 1 ESC (used with DShot 1200).

Oko001, 2019-04-20:
I use 10.5.1 configurator at now, but it was the same with the previous version too.

etracer65, 2019-04-20:
As requested when you opened the issue, please provide a `diff` of your settings when you have the problem. Without that we can't help.

Oko001, 2019-04-20:
Sorry, I missed that.
I reflashed the 4.0, and without UART setting, just calibrated the accelerometer, then went to CLI for diff, and after reboot the accelerometer and gyro disappeared.
# diff

# version
# Betaflight / MATEKF722SE (MF7S) 4.0.0 Apr 11 2019 / 07:26:45 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name MATEKF722SE
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_calibration = 80,-95,392

# profile
profile 0

# rateprofile
rateprofile 0

# end the command batch
batch end

#

Oko001, 2019-04-20:
Windows 10 if it counts..

etracer65, 2019-04-20:
I just tested with that board and can't replicate any problems. It's always possible you have a board with an intermittent or faulty gyro. As a test, try using the second gyro on that board to see if you can still replicate the problem.
```
set gyro_to_use = SECOND
save
```

Oko001, 2019-04-20:
The problem is still there with the second gyro. Flashed back to 3.5.7 or 3.5.6 or 3.5.5. everything is fine.

etracer65, 2019-04-20:
Using the same board I can't replicate any problems. I also know that this board is quite popular and many people are using it without problems (including some of the other devs).

Oko001, 2019-04-20:
So this is a faulty FC. Strange that it has no issues with the older FW. Thank You the help.

etracer65, 2019-04-20:
Well I didn't exactly say that. Just that I've been unable to replicate and there have been no other reports of problems. Could be a faulty flight controller, or possibly some bug that you're able to trigger that nobody else is seeing. Just to be absolutely sure, can you document the steps you can use to reproduce? Start with flashing the current 4.0 release using a "full-chip erase" and then detail the steps you follow.

Oko001, 2019-04-20:
Ok. After flashing the 4.0 release, at first connect I have accelerometer. I can calibrate it. Doesn't matter that I calibrate it or not, set up anything or not, after the first reboot the Gyro/Accelerometer is not available. Just if I reflash the firmware - but with the same sympthom. Or flash back the older ones, wich are free from problems.

laz2wiringpi, 2019-04-21:
try this its a long shot .
remove props 
plug in battery then connect usb ..
what happens ?

Oko001, 2019-04-21:
At first try, accelerometer was found. Then I saved the 8KHz/8KHz setting, and never again with or without battery.

laz2wiringpi, 2019-04-21:
i have 2 of these boards , one in use and the other i use for testing.. unable to reproduce like the others. 
 bf 4 has more in it so it just doesn't happen yet with that firmware and its a defective board.
as a wild guess can you try 
https://ci.betaflight.tech/job/Betaflight/1552/artifact/obj/
don't use in flight just see if the problem is still there ?

Oko001, 2019-04-21:
I tried it. Same symptom. At first connect ACC is live, after the first reboot it disappears. Only with 4.0 release and release candidate versions. 
 It is strange for me. I thought that a bad gyro will be bad with all software. At least I now that noone else has the same problem, so my FC is defective.

laz2wiringpi, 2019-04-21:
most likely a bad board  just strange it works in bf 3.x   could be some undiscovered boot or gyro timing init issue . 
look over the board both sides  i  use a microscope .   and lots of times i can fix it with a cleaning. other times is just a component that is just barely reconnected.

Oko001, 2019-04-21:
I hesitated on take apart the quad, but I did. And found the problem - wich wasn't there . After the save/reboot I got error messages:
2019-04-21 @ 15:28:33 -- EEPROM saved
2019-04-21 @ 15:28:33 -- Device - Rebooting
2019-04-21 @ 15:28:33 -- Unrecoverable failure of serial connection, disconnecting...
2019-04-21 @ 15:28:33 -- Failed to close serial port

After that, it finds again the COM3 port, and connect, and seems to be ok, but without acc and gyro.
If I disconnect the USB, reconnect it again, it works!
So I have to reconnect USB after every save, what is not needed at previous versions.
And not needed on my another FC boards (at least with Airbot F4 with BF 4.0) with these STM/Zadig/CP210x drivers. 
Any solution for this?

laz2wiringpi, 2019-04-21:
sounds like board has an issue .  as i cannot reproduce . pull down  resistor or open/shorted  diode  ?
dont  i have  seen a schematic for  Matek's boards.  I have been meaning to ask them for one but not likely to get one.  

could be you usb  .. have you tried another computer.?

Oko001, 2019-04-21:
At least, I can configure it now at this way, and gyro is ok. Most important that I will be able to fly with it.

I have no another computer nearby, but at same port, with same USB cable, same drivers, another (F4) flight controller works well with BF 4.0 too.
Is there some difference between BF 3. versions and BF 4.0 serial port or USB handling?  Might my FC has problem with the last. I cannot see any physical damage on it. Maybe there is, but a damaged diode or a resistor shouldn't work with previous FW-s right? I've been flashed many times back and forth with the same result, so it is not a typical changing contact problem.

 Thank you etracer65 and laz2wiringpi for the help!

 Let'me leave open the question for a while! I hope that the solution is exists. :)

laz2wiringpi, 2019-04-21:
Maybe there is, but a damaged diode or a resistor shouldn't work with previous FW-s right?

no ..   the time to pull the line low/high. it could even be in the cpu .
bf 4.0 not 3.6 for a reason... a lot was done. 
 
 i had an issue one time drove me nuts random issues with my computer do to a https://en.wikipedia.org/wiki/Whisker_(metallurgy) .
now it may come up with others then it should be investigated but for this issue Im satisfied it's a bad board.  if you replace it . send me the old one .  i can always use the parts . 👀

Oko001, 2019-04-22:
Summary:
 The FC has flown about 10 packs alltogether. 1 time with a strange glitch, and 1 short time disarm under first 2 packs. I thought that it happened because of vibrations, and I have to check the boards mount. 
 Then I would like to try the new BF features. The board can't reboot with BF 4.0, I have to reconnect the USB cable or I loose the gyro. With this method, I can configure it. I had only a 3 min test flight with it with stock PID and filtering. Only low speeds at night. It is smooth as I imagined with 6", 5s, Runcam3, 650g AUW.
 
 Possible that I have transistor failure at the cpu, or any other hardware problem. Time will tell. If I will not have new problem, I will keep this board.

  Thank you for the help again!

closed 2019-04-22 by Oko001

brainbubblersbest, 2019-04-23:
Hello, just want to reply i have same Problem with my f722se. Allmost all Sensors disappear from Sensors tab after configure gps from auto to qmc5338. Set it back to auto and everything is fine. That makes it less urgent for me. Because it worked flawless on Auto. If needed, I can attach a diff with described settings.

laz2wiringpi, 2019-04-23:
not the same problem

imbradas, 2020-07-07:
Hi, I had same issue with gyro with the same board. I turns out the mistake was done when reflashing the board. I have chosen "Matek f722" in configurator instead of "Matek f722se". Once reflashed with correct software gyro came back. Even on 4.2 version of betaflight. Hope it helps.

NYVINCHENZ, 2020-08-09:
A friend of mine mentioned a flight controller program to emuflight i have a matek f722-std  with the firmware 4.2 version i lose my gyro also is there something wrong with that firmware update is the f722 se that same ?