betaflight/betaflight #8056: Serial broken after flashing 4.0.0

richard-scott, 2019-04-20:
**Describe the bug**
Serial port stops working for the configurator. Maybe its a repeat of https://github.com/betaflight/betaflight/issues/5502

**To Reproduce**

1. Steps to reproduce the behavior:
2. Flash BF 4.x to an AnyFC F7.
3. Disconnect USB
4. Connect USB and click Connect in BFC.

**Expected behavior**
Serial port should work with BFC.

**Flight controller configuration**
None.

etracer65, 2019-04-20:
Please update the title to be related to AnyFC F7. There are no reports of problems with other F7 boards (or any other flight controllers) at this time. I just tested several other F7 boards and they had no problems.

neverknownbetter, 2019-04-20:
Same issue with omnibus F4 no soft serial wor5

etracer65, 2019-04-20:
@neverknownbetter 
> Same issue with omnibus F4 no soft serial working

You have reported a completely different problem related to ESC telemetry and it's not at all related to this issue.

richard-scott, 2019-05-24:
Has there been any changes on this? Do I need to re-test?

richard-scott, 2019-06-16:
I've just tried BF 4.1.0 build [#1700](https://ci.betaflight.tech/job/Betaflight/1700/) and I get this when I click connect.

```
2019-06-16 @ 22:10:02 -- Serial port successfully opened with ID: 5
2019-06-16 @ 22:10:02 -- MultiWii API version: 1.42.0
2019-06-16 @ 22:10:02 -- Flight controller info, identifier: BTFL, version: 4.1.0
2019-06-16 @ 22:10:02 -- Running firmware released on: Jun 16 2019 00:47:02
2019-06-16 @ 22:10:12 -- No configuration received within 10 seconds, communication failed
2019-06-16 @ 22:10:12 -- Serial port successfully closed
```

mikeller, 2019-06-17:
@richard-scott: Looks like something is locking up with the ANYFCF7. Unfortunately I do not have an engineering sample of this board. Maybe @sambas can help?

richard-scott, 2019-08-21:
Seems ok with  4.1.0 #1824

closed 2019-08-21 by richard-scott