betaflight/betaflight #8057: OMNIBUSF4SD esc sensor an soft serail

neverknownbetter, 2019-04-20:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

neverknownbetter, 2019-04-20:
beta flight 4.0. unable to activate esc sensor and enable cpu serial ports in ports tab. target hex OMNIBUSF4SD. worked with 3.5 now no current sensing available.

etracer65, 2019-04-20:
Please follow the directions in the instructions that you failed to read the instructions about removing.

neverknownbetter, 2019-04-20:
Which part of the instruction informs about making esc amp telementry work different to betaflight 3.5?

etracer65, 2019-04-20:
Provide the requested information.

> **Flight controller configuration**
> Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: [...]

jflyper, 2019-04-20:
@neverknownbetter You should supply information requested in the pre-formatted text when you first opened this issue ticket (the text is still there in the first comment). What did you think they are for?

neverknownbetter, 2019-04-21:
@etracer65  
```# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Apr 11 2019 / 07:31:36 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name OMNIBUSF4SD
manufacturer_id 

# name

# resources
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 A02
resource MOTOR 4 B01

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature MOTOR_STOP
feature SOFTSERIAL
feature ESC_SENSOR

# beeper
beeper -GPS_STATUS
beeper -READY_BEEP
beeper -ON_USB
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
beeper -RC_SMOOTHING_INIT_FAIL

# beacon
beacon RX_SET

# map

# serial
serial 0 1024 115200 57600 0 115200
serial 2 64 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 1 1650 2100 0 0
aux 1 2 3 900 1250 0 0
aux 2 13 2 1600 2100 0 0
aux 3 28 3 1350 2100 0 0
aux 4 35 4 1700 2100 0 0
aux 5 49 3 1700 2100 0 0

# adjrange
adjrange 0 2 0 900 1275 12 0 0 0
adjrange 1 1 0 1300 1700 12 0 0 0
adjrange 2 0 0 1725 2100 12 0 0 0
adjrange 3 0 5 900 1300 29 5 0 0
adjrange 4 1 5 1300 1700 29 5 0 0

# rxrange

# vtx

# rxfail
rxfail 6 s 1900
rxfail 7 s 1000

# display_name

# master
set acc_calibration = 0,-16,-23
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1020
set max_check = 1980
set rssi_channel = 11
set fpv_mix_degrees = 25
set serialrx_provider = SBUS
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT1200
set align_board_yaw = 270
set bat_capacity = 1500
set current_meter = ESC
set vbat_scale = 112
set ibata_scale = 250
set ibata_offset = 10
set small_angle = 180
set pid_process_denom = 1
set osd_cap_alarm = 1100
set osd_vbat_pos = 6636
set osd_rssi_pos = 6145
set osd_tim_1_pos = 54
set osd_tim_2_pos = 6647
set osd_flymode_pos = 57
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 2528
set osd_mah_drawn_pos = 6594
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 6170
set osd_warnings_pos = 14602
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 488
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 481
set osd_stick_overlay_left_pos = 4353
set osd_stick_overlay_right_pos = 4374
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_rpm = ON
set vtx_channel = 4
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5800

# profile
profile 1

set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 200
set vbat_pid_gain = ON
set feedforward_transition = 30
set iterm_relax = RPY
set iterm_relax_type = GYRO
set i_pitch = 85
set i_roll = 80
set d_roll = 30
set f_yaw = 35
set d_min_roll = 25
set d_min_pitch = 28
set d_min_boost_gain = 30
set d_min_advance = 50

# rateprofile
rateprofile 1

set roll_rc_rate = 110
set pitch_rc_rate = 110
set yaw_rc_rate = 120
set roll_expo = 10
set pitch_expo = 10
set roll_srate = 75
set pitch_srate = 77
set yaw_srate = 80

# end the command batch
batch end```

etracer65, 2019-04-21:
Can you please clearly describe the problem? Also again as initially requested, please provide the hardware details and how you have them connected. For example, what ESCs are you using? What's the firmware version on those ESC's. You seem like you're trying to use ESC telemetry - how do you have the telemetry signal from the ESC's connected to your flight controller?

We can't help by guessing.

neverknownbetter, 2019-04-21:
Sorry about the lack of info provided. New to using git hub. My esc's are connected to rx1 on the flight controller.  Please see below a copy of BLHeli suite 32 settings.

[image]

neverknownbetter, 2019-04-21:
sorry again. Yes the problem is reading current from esc's. There also is no extra ports with feature soft serail.

jflyper, 2019-04-23:
@neverknownbetter 
I've tested this with Omnibus F4 V3 (OMNIBUSF4SD target) freshly flashed with 4.0.0 and loaded with your config diff. ESCs are Airbot Wraith 32s (I suspect it is identical to Betaflight BLHeli32) and are connected to RX1.

It worked straight without a problem.

I suspect you have some hardware oriented issue. Since you are not providing information about the hardware, I'm closing this.

closed 2019-04-23 by jflyper