betaflight/betaflight #8060: rpmfilter: one motor output not working on SPEEDYBEEF4 (SBF4)

elim99, 2019-04-21:
Trying to get another quad working with rpmfilter. 

MOTOR 1 does not spin. 
~~~
# Betaflight / SPEEDYBEEF4 (SBF4) 4.0.0 Apr 11 2019 / 07:40:33 (9ad2cc844) MSP API: 1.41
~~~
Here is status:
~~~
Dshot reads: 1230
Dshot invalid pkts: 0
Dshot irq micros: 3
Dshot RPM Motor 0: 4095
Dshot RPM Motor 1: 0
Dshot RPM Motor 2: 0
Dshot RPM Motor 3: 0
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 
~~~
DMA:
~~~
# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma pin A03 0
# pin A03: DMA1 Stream 1 Channel 6
dma pin C06 0
# pin C06: DMA2 Stream 2 Channel 0
dma pin C07 1
# pin C07: DMA2 Stream 3 Channel 7
dma pin C08 1
# pin C08: DMA2 Stream 4 Channel 7
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin A15 0
# pin A15: DMA1 Stream 5 Channel 3
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin B08 0
# pin B08: DMA1 Stream 7 Channel 2
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2
~~~
And resources:
~~~
# resource
resource BEEPER 1 C13
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 C08
resource MOTOR 4 C09
resource MOTOR 5 A15
resource MOTOR 6 A08
resource MOTOR 7 B08
resource PPM 1 A03
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_RX 1 A10
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource I2C_SCL 1 B06
resource I2C_SDA 1 B07
resource LED 1 B09
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 B03
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 B04
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 B05
resource ESCSERIAL 1 A03
resource ADC_BATT 1 C02
resource ADC_RSSI 1 C03
resource ADC_CURR 1 C01
resource PINIO 1 B00
resource FLASH_CS 1 C00
resource OSD_CS 1 B10
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 B11
resource USB_DETECT 1 B12
~~~

Flavoredcrayon, 2019-04-25:
Also getting a problem with motor one. She is reading the same rpm on Motor 0 as well. The esc would also  replay the last two startup tones every 10 or so seconds.

Flavoredcrayon, 2019-04-26:
So I figured out a solution. I changed the motor one dma to whatever was at index one. This is what I typed in the cli.

dma pin c06 1

Motors spin:
[image]

I barely know what a dma is much less what this can hurt/conflict with so I would wait for someone who knows what they are talking about to chime in before trying. Here is some info for that someone to see if it mucked anything up.

timer A03 1
timer C06 1
timer C07 1
timer C08 1
timer C09 1
timer A15 0
timer A08 0
timer B08 0
timer B06 0

# dma
dma ADC 1 0
# ADC 1: DMA2 Stream 0 Channel 0
dma pin A03 0
# pin A03: DMA1 Stream 1 Channel 6
dma pin C06 1
# pin C06: DMA2 Stream 2 Channel 7
dma pin C07 1
# pin C07: DMA2 Stream 3 Channel 7
dma pin C08 1
# pin C08: DMA2 Stream 4 Channel 7
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin A15 0
# pin A15: DMA1 Stream 5 Channel 3
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin B08 0
# pin B08: DMA1 Stream 7 Channel 2
dma pin B06 0
# pin B06: DMA1 Stream 0 Channel 2

elim99, 2019-04-27:
> dma pin c06 1

Thank you. Workaround seems to work for me.