betaflight/betaflight #8061: Betaflight 4.0 LUA script for GPS settings no working

michael-blue2, 2019-04-21:
Betaflight/Gps Rescue/ Page on lua scripts is not working/loading
Betaflight 4.0 matek f405
Even with satellites its still the same

laz2wiringpi, 2019-04-21:
https://github.com/betaflight/betaflight-tx-lua-scripts/releases
is not done for 4.0 i think its still a work in progress 

https://ci.betaflight.tech/job/Betaflight%20Lua%20-%20X7,%20X9,%20X9D,%20X9D+,%20X10,%20X12/
i have  used build #33 and it works ( beta versions)

mikeller, 2019-04-28:
@laz2wiringpi is correct, the LUA script to work with Betaflight 4.0 has still to be released - there were improvements that were added until right about now.

michael-blue2, 2019-04-28:
At first when i saw that some months ago on the lua, i thought thats it was ready and only the bf was needed an update. All good then, Thank you both guys!

closed 2019-04-28 by michael-blue2