betaflight/betaflight #8062: add possibility to adjust ascend speed in gps resque mode

m013, 2019-04-21:
a quad just goes into full throttle in gps rescue event and since we have got this strange logic when it  ascends to the highest altitude reached in a flight instead of preconfigured return altitude (gps_rescue_initial_alt), it can kill battery very quickly

we should have a possibility to configure ascend speed, like rth velocity (gps_rescue_ground_speed)

Dom4n, 2019-04-21:
You can set throttleMax to lower value, then it will not use 100% of throttle while ascending. Another option is to adjust throttle PID so it will not go full speed immediately.

m013, 2019-04-21:
@Dom4n if i use throttleMax will it affect return speed?

IllusionFpv, 2019-04-21:
https://github.com/betaflight/betaflight/pull/8015

closed 2019-05-28 by stale[bot]