betaflight/betaflight #8065: When can you customize template generation

liguo-git, 2019-04-22:
Many things are not needed, such as more receiver protocols, motor protocols, GPS protocols, etc.

Docteh, 2019-04-22:
This looks more like a conversation starter than an actual issue. What sort of templates are you referring to?

watageek, 2019-04-22:
You can already do this 

# define or //
Or
#Undef

You can easely build a binairy with/ without features...

I think you complain about an issue that is not an issue...

swilkens, 2019-04-22:
@liguo-git 

If I understood you correctly, you want to understand how to enable / disable features for a specific target?

In general:

`src/main/target/common_pre.h`

This file configures the common features, it also configures which features are enabled / disabled in a general sense - based on three things:

1. The processor type (F1 / F3 / F4 / F7...)
2. The available flash space (64, 128, 256...)
3. The `FEATURE_CUT_LEVEL` defined in the target's `target.mk` file

The best way of making modifications is by editing the target's `target.h` file. For example:

`src/main/target/BETAFLIGHTF3/target.h`

If, for example I want to add the `USE_PWM` feature to this `BETAFLIGHTF3` target, I can simply add:

`#define USE_PWM`

to the file.

If I want to create ore flash space by removing some protocols, I can add the following to disable support for the Spektrum serial protocol:

`#undef USE_SERIALRX_SPEKTRUM`

Does this answer your question?

closed 2019-05-29 by stale[bot]