betaflight/betaflight #8066: OMNIBUSF4 can use 2 motors at most with 4.0.0

andrejpodzimek, 2019-04-22:
`OMNIBUSF4` ([HGLRC stack](https://www.hglrc.com/collections/flight-controller/products/hglrc-xjb-f44032-f4-flight-control-d-shot-version-40a-bl32_40a-3-5s-esc-blheli_32)) is unusable with Betaflight 4.0.0, because only 2 motors work.

```
# version
# Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Apr 22 2019 / 03:13:32 (689882bc8) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4
manufacturer_id AIRB
mcu_id 002000263037470535363837
signature 

# name
name Ejx

# resources
resource MOTOR 1 B01
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 A03
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_TX 11 A00
resource SERIAL_RX 3 NONE

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -AIRMODE
feature -RX_SPI
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY
feature LED_STRIP
feature ESC_SENSOR

# beeper

# beacon

# map
map AETR1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led
led 0 7,7:SD:ATW:0

# color

# mode_color

# aux
aux 0 0 1 1600 2100 0 0
aux 1 28 4 1600 2100 0 0
aux 2 4 1 900 2100 0 0
aux 3 1 0 900 1300 0 0
aux 4 2 0 1300 1700 0 0
aux 5 13 3 1200 2100 0 0

# adjrange
adjrange 0 0 2 900 2100 12 2 0 0

# rxrange

# vtx

# rxfail
rxfail 4 s 1000
rxfail 7 s 2000

# display_name

# master
set gyro_lowpass_type = PT1
set acc_calibration = -128,87,139
set mag_hardware = NONE
set baro_hardware = NONE
set max_check = 1950
set rssi_channel = 16
set serialrx_provider = SBUS
set blackbox_p_ratio = 0
set blackbox_device = NONE
set dshot_idle_value = 200
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 24000
set align_board_yaw = 90
set bat_capacity = 600
set current_meter = ESC
set battery_meter = ESC
set pid_process_denom = 1
set frsky_unit = METRIC
set osd_vbat_pos = 2433
set osd_flymode_pos = 2477
set osd_throttle_pos = 2520
set osd_vtx_channel_pos = 2466
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_current_pos = 2502
set osd_mah_drawn_pos = 2497
set osd_power_pos = 2512
set osd_esc_rpm_pos = 2392
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_channel = 8
set vtx_low_power_disarm = ON
set vtx_freq = 5880
set vcd_video_system = PAL

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass2_type = PT1
set vbat_pid_gain = ON
set anti_gravity_gain = 2000
set iterm_rotation = ON
set smart_feedforward = ON
set iterm_relax = RPY
set iterm_relax_type = GYRO
set yaw_lowpass_hz = 100
set p_pitch = 50
set i_pitch = 25
set d_pitch = 15
set p_roll = 45
set i_roll = 25
set d_roll = 15
set p_yaw = 50
set i_yaw = 30
set horizon_transition = 80
set d_min_roll = 0
set d_min_pitch = 0

# profile
profile 1

set dterm_lowpass_type = PT1
set vbat_pid_gain = ON
set anti_gravity_gain = 2000

# profile
profile 2

set dterm_lowpass_type = PT1
set vbat_pid_gain = ON
set anti_gravity_gain = 2000

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 80

# rateprofile
rateprofile 2

set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 80

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 1
```

I tried a number of different configurations, both release binaries and builds, I also tried to port [my hacks for 3.3.3](https://github.com/betaflight/betaflight/issues/6026) into Betaflight 4.0.0, but no, it simply doesn't work.

Surprisingly enough, when I apply my hacks, motors 2 and 4 are working. Without my hacks, motors 1 and 3 are working.

I tried a number of things:
*  `dshot_burst` (no effect)
* timer and dma configurations (setting the magic numbers between 0/1/2) (no effect, apart from rendering the two remaining motors unusable)
* multishot (no effect; 2 motors are usable, the remaining two wobble randomly with no reponse to the motors tab input)
* dshot 600 / 1200 (which I use by default) (no effect, 2 motors work)

The two working motors have ESC telemetry shown in the Power and Battery configurator section, unlike those that don't work.

andrejpodzimek, 2019-04-22:
Gah. :disappointed: It works with an unmodified configuration. There must be a resource conflict of some sort that hadn't been there before.

andrejpodzimek, 2019-04-22:
It's softserial. After some poking around, I see that it conflicts with the two motors for some reason.

andrejpodzimek, 2019-04-22:
Tested with 3.5.7 and everything works fine there. I have DShot 1200 on all 4 motors, the LED works and also the SmartPort on softserial works; I have telemetry. So this definitely looks like a regression in 4.0.0.

It might be somehow related to the way 4.0.0 handles timers and DMA. I was impressed how it somehow automatically figured out a timer and DMA assignment for my non-standard LED connection. The programmable LED just started working without any of the timer hacks needed in previous versions.

The downside is that using softserial disables two out of four motors (which is not specific to DShot and `dshot_burst` doesn't influence it either). The two motors that don't work don't do anything under DShot and wobble under multishot. Surprisingly, increasing the throttle with multishot only influences the working motors while the wobbly ones keep wobbling the same way, with no sign of a throttle increase.

This build wouldn't work with stock 3.5.7 because of a few missing timer definitions, so I applied my hacks here as well.

```diff
diff --git a/src/main/target/OMNIBUSF4/target.c b/src/main/target/OMNIBUSF4/target.c
index 129cd41bf..b0bc62a17 100644
--- a/src/main/target/OMNIBUSF4/target.c
+++ b/src/main/target/OMNIBUSF4/target.c
@@ -48,12 +48,15 @@ const timerHardware_t timerHardware[USABLE_TIMER_CHANNEL_COUNT] = {
     DEF_TIM(TIM2,  CH3, PA2,  TIM_USE_MOTOR,               0, 0), // S4_OUT D1_ST1
 
 #if defined(OMNIBUSF4SD) || defined(EXUAVF4PRO)
-    DEF_TIM(TIM5,  CH2, PA1,  TIM_USE_MOTOR,               0, 0), // S5_OUT
+    DEF_TIM(TIM5,  CH1, PA0,  TIM_USE_NONE,                0, 0), // SoftSerial
+    DEF_TIM(TIM2,  CH2, PA1,  TIM_USE_MOTOR,               0, 0), // S5_OUT
     DEF_TIM(TIM4,  CH1, PB6,  TIM_USE_LED,                 0, 0), // LED strip for F4 V2 / F4-Pro-0X and later (RCD_CS for F4)
 #elif defined(SYNERGYF4)
     DEF_TIM(TIM5,  CH2, PA1,  TIM_USE_LED,                 0, 0), // LED strip
 #else
-    DEF_TIM(TIM5,  CH2, PA1,  TIM_USE_MOTOR | TIM_USE_LED, 0, 0), // S5_OUT
+    DEF_TIM(TIM5,  CH1, PA0,  TIM_USE_NONE,                0, 0), // SoftSerial
+    DEF_TIM(TIM2,  CH2, PA1,  TIM_USE_MOTOR,               0, 0), // S5_OUT
+    DEF_TIM(TIM3,  CH1, PC6,  TIM_USE_LED,                 0, 0), // TX6 -> LED
 #endif
 #if defined(SYNERGYF4)
     DEF_TIM(TIM1,  CH1, PA8,  TIM_USE_ANY,                 0, 0), // CAM_CTL
```

jflyper, 2019-04-22:
@andrejpodzimek Is there any reason to keep this open as this is likely to be your "_*how-to-hack*_" ?

P.S. I think you can do all this with unified target without building a custom target.

andrejpodzimek, 2019-04-22:
It looks like a bug to me, because something that worked with 3.5.7 doesn’t work with 4.0.0 any more. I could not get motors *and* softserial to work with 4.0.0.

Is there a howto about the unified targets? Maybe the answer is in the new `dma` and `timer` commands, but thus far I couldn’t find a way to resolve this in 4.0.0. Softserial on —> 2 motors gone.

jflyper, 2019-04-22:
> It looks like a bug to me, because something that worked with 3.5.7 doesn’t work with 4.0.0 any more. I could not get motors and softserial to work with 4.0.0.

I'm confused.

You have a craft with non-standard 3.5.7 based custom firmware working, and claiming there is a bug in 4.0.0 because the same custom mod doesn't work?

andrejpodzimek, 2019-04-22:
> You have a craft with non-standard 3.5.7 based custom firmware working, and claiming there is a bug in 4.0.0 because the same custom mod doesn't work?

No, I’m claiming that I can reproduce the problem also on stock, unmodified Betaflight, by setting softserial 1 TX to `PA0`. Only two motors work then (on stock 4.0.0, as downloaded by the configurator). Other 2 motors don’t work.

(Application of my hack changes *which* of the two motors work, but there’s still only 2 of them on 4.0.0 with softserial on `PA0`.)

On 3.5.7, (1) all motors always work and (2) only softserial depends on the hack to work; no motors affected.

jflyper, 2019-04-22:
@andrejpodzimek Ah, okay, so the symptom suggest that assigning non-existent timer pin PA0 to softserial causes motors on TIM2 to stop working. Lemme take a look.

andrejpodzimek, 2019-04-22:
> @andrejpodzimek Ah, okay, so the symptom suggest that assigning non-existent timer pin PA0 to softserial causes motors on TIM2 to stop working. Lemme take a look.

Thanks for taking a look. BTW, here's the quickest way to reproduce the problem:

1. Flash stock `OMNIBUSF4` Betaflight (as downloaded by the configurator), version 4.0.0.
2. Do this in the CLI:
    ```
    set motor_pwm_protocol = DSHOT1200
    feature SOFTSERIAL
    feature TELEMETRY
    resource RX_SPI_EXTI 1 NONE
    resource SERIAL_TX 11 A00
    serial 30 32 115200 57600 0 115200
    save
    ```
3. Connect the battery, go to the Motors tab and try to increase Master. Only 2 motors will spin.

The `DSHOT1200` is not necessary to expose the error; it fails with `MULTISHOT` also. But the inoperable motors wobble in a strange way with Multishot and I didn't want them or the ESCs to get burned.

Now I see that I should have posted this MWE at the very beginning, but got so overwhelmed by an all-nighter of trial and error that I could no longer reason clearly. Sorry about that.

jflyper, 2019-04-22:
Thanks, but I now know what's going on ;)

froqstar, 2019-04-24:
Similar problem with Mamba F405 Mini (FuryF4OSD Target):
When I set softserial 1 TX to TX6 pin (C06) and enable smartport telemetry on softserial 1, motor 2 stops working. Disabling smartport makes motor 2 work normally again.

jflyper, 2019-04-25:
@froqstar 

In your case, the base cause of the symptom is the same as what @andrejpodzimek revealed, but the issue is inherent to the FURYF4OSD target, as the target is not configured to use PC6 as a timer pin.
You will not be able to do what you want to do even if the bug revealed here is fixed without using unified target.

mikeller, 2019-04-25:
@andrejpodzimek, @froqstar: Even with non-unified targets in the current release you should be able to work around this by using `timer <pin> <option>` to configure a timer for PA0 (OMNIBUSF4) or PC6 (FURYF4OSD) respectively.

andrejpodzimek, 2019-04-27:
As for `timer <pin> <option>`, I have no clue how to use that. Tried ~20 combinations of those magic numbers for the 4 motor pins, but the only state change I managed to cause was to make 3 motors inoperable rather than 2.

I also cherry-picked the #8093, but it didn't change anything; the problem was exactly the same. (So I guess it just fixes the obvious inconsistency spotted by @jflyper, but may not be a complete fix to this issue.)

I'll try the `STM32F405` target [based on the documentation](https://github.com/betaflight/betaflight/blob/master/unified_targets/docs/CreatingAUnifiedTarget.md), to see if it gets any different with the unified target. But I have no idea what the `timer` command is supposed to do there.

andrejpodzimek, 2019-04-27:
Wow! The unified target **just** **works**. My drone is back in the air with telemetry on softserial and also LEDs. Here's a diff:

```
# version
# Betaflight / STM32F405 (S405) 4.1.0 Apr 27 2019 / 23:47:51 (3a58865c9) MSP API: 1.42

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name OMNIBUSF4
manufacturer_id AIRB
mcu_id 002000263037470535363837
signature 

# name

# resources
resource BEEPER 1 B04
resource MOTOR 1 B01
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 A03
resource MOTOR 5 A01
resource MOTOR 6 A08
resource PPM 1 B14
resource SONAR_TRIGGER 1 A01
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 C06
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 11 A00
resource SERIAL_RX 1 A10
resource SERIAL_RX 6 C07
resource INVERTER 1 C00
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B14
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource FLASH_CS 1 B03
resource OSD_CS 1 A15
resource RX_SPI_CS 1 D02
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05

# timer
timer B14 2
timer B15 2
timer C06 1
timer C07 1
timer C08 1
timer C09 1
timer B00 1
timer B01 1
timer A03 0
timer A02 0
timer A01 1
timer A08 0
timer A09 0
timer A10 0
timer A00 1

# dma
dma ADC 2 1
# ADC 2: DMA2 Stream 3 Channel 1
dma pin C06 0
# pin C06: DMA2 Stream 2 Channel 0
dma pin C07 0
# pin C07: DMA2 Stream 2 Channel 0
dma pin C08 0
# pin C08: DMA2 Stream 2 Channel 0
dma pin C09 0
# pin C09: DMA2 Stream 7 Channel 7
dma pin B00 0
# pin B00: DMA1 Stream 7 Channel 5
dma pin B01 0
# pin B01: DMA1 Stream 2 Channel 5
dma pin A03 1
# pin A03: DMA1 Stream 6 Channel 3
dma pin A02 0
# pin A02: DMA1 Stream 1 Channel 3
dma pin A01 0
# pin A01: DMA1 Stream 4 Channel 6
dma pin A08 0
# pin A08: DMA2 Stream 6 Channel 0
dma pin A09 0
# pin A09: DMA2 Stream 6 Channel 0
dma pin A10 0
# pin A10: DMA2 Stream 6 Channel 0

# feature
feature -RX_PARALLEL_PWM
feature -AIRMODE
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY
feature LED_STRIP
feature OSD
feature ESC_SENSOR

# serial
serial 0 64 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led
led 0 7,7::AT:0

# aux
aux 0 0 1 1600 2100 0 0
aux 1 28 4 1600 2100 0 0
aux 2 4 1 900 2100 0 0
aux 3 1 0 900 1300 0 0
aux 4 2 0 1300 1700 0 0
aux 5 13 3 1200 2100 0 0

# adjrange
adjrange 0 0 2 900 2100 12 2 0 0

# master
set gyro_lowpass_type = PT1
set acc_calibration = -36,84,151
set mag_bustype = I2C
set mag_i2c_device = 2
set mag_hardware = NONE
set baro_bustype = I2C
set baro_i2c_device = 2
set baro_hardware = NONE
set max_check = 1950
set rssi_channel = 16
set serialrx_provider = SBUS
set rx_spi_bus = 3
set adc_device = 2
set blackbox_p_ratio = 0
set blackbox_device = NONE
set min_throttle = 1070
set dshot_idle_value = 200
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set align_board_yaw = 90
set bat_capacity = 600
set current_meter = NONE
set battery_meter = ESC
set beeper_inversion = ON
set beeper_od = OFF
set pid_process_denom = 1
set frsky_unit = METRIC
set system_hse_mhz = 8
set vtx_channel = 8
set vtx_low_power_disarm = ON
set vtx_freq = 5880
set vcd_video_system = PAL
set max7456_clock = FULL
set max7456_spi_bus = 3
set dashboard_i2c_bus = 2
set flash_spi_bus = 3
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW180
set gyro_2_spibus = 1
```

Noteworthy stuff: `timer A00 1`

This^^^ was needed to get softserial going. The timer is configured automatically for LEDs (`timer C06 1`) when the `LED_STRIP` resource is changed. However, this doesn't happen automatically for softserial (`timer A00 1`).

Exact steps taken in this case:
* Check out current `master` (078b25c565e6d9b4151e539bca0b5efdabfeee80). I had to take master, because `STM32F405` has a build failure in `4.0.0`.
* Cherry-pick #8093. (I don't know if it's necessary or if the underlying issue can be circumvented otherwise. But just in case.)
  ```bash
  git remote add mikeller git@github.com:mikeller/betaflight.git
  git fetch mikeller
  git cherry-pick e8427be520236cb7165fd34963703395c593b3a1
  git remote remove mikeller  # optional
  ```
* Build `OMNIBUSF4` and `STM32F405`.
* Follow the [unified target howto](https://github.com/betaflight/betaflight/blob/master/unified_targets/docs/CreatingAUnifiedTarget.md).
* Restore the 3.5.7 (or watever old) configuration, skipping softserial + smartport.
* Configure softserial and smartport. There won't be any telemetry.
* Set `timer A00 1`. Telemetry will start working.

So everything works fine for me at the moment. The unified targets look like a great improvement; I won't have to maintain timer hacks for my non-standard builds any more.

PR #8093 (2019-04-27): Fixed assignment of random timer for pins without configured timer option.
Fixes #8066, #8071.
closed 2019-04-27 by mikeller