betaflight/betaflight #8067: BLHeli Passthrough not working with BF 4.0.0 Colibri Race

djmartin, 2019-04-22:
**Describe the bug**
When connecting BLHeliSuite via Passthrough - will not connect to ESCs since upgrading to BF 4.0.0

**To Reproduce**
Connect FC to USB, confirm COM port connects, but when starting BLHeliSuite & then connecting - passthrough will not work

**Expected behavior**
Be able to read BLHeli values & configurations

**Flight controller configuration**
diff all
# Betaflight / COLIBRI_RACE (CLBR) 4.0.0 Apr 11 2019 / 07:03:08 (9ad2cc844) MSP API: 1.41 / FEATURE CUT LEVEL 1
batch start
defaults nosave
mcu_id 003100244733570520303136

feature -LED_STRIP
feature -AIRMODE
feature TELEMETRY
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE
aux 0 0 1 1275 2100 0 0
aux 1 1 0 900 1200 0 0
aux 2 2 0 1300 1700 0 0
aux 3 13 3 1800 2100 0 0
aux 4 28 0 1700 2100 0 0
adjrange 0 0 4 900 1400 12 4 0 0
adjrange 1 1 4 1400 1800 12 4 0 0
adjrange 2 2 4 1800 2100 12 4 0 0
rxfail 6 s 1000
set gyro_lowpass_type = PT1
set dyn_notch_min_hz = 100
set dyn_lpf_gyro_min_hz = 100
set dyn_lpf_gyro_max_hz = 300
set acc_calibration = -26,15,71
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1100
set rssi_channel = 7
set max_aux_channels = 5
set serialrx_provider = CRSF
set airmode_start_throttle_percent = 32
set blackbox_device = NONE
set min_throttle = 1060
set max_throttle = 1990
set use_unsynced_pwm = OFF
set motor_pwm_protocol = ONESHOT125
set motor_pwm_rate = 2000
set failsafe_delay = 20
set failsafe_off_delay = 40
set failsafe_throttle = 1080
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1800
set vbat_max_cell_voltage = 431
set vbat_warning_cell_voltage = 340
set battery_meter = ADC
set small_angle = 180
set deadband = 2
set yaw_deadband = 3
profile 0

set dyn_lpf_dterm_min_hz = 100
set dyn_lpf_dterm_max_hz = 200
set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set anti_gravity_gain = 10000
set iterm_relax = RPY
set iterm_relax_type = GYRO
set iterm_relax_cutoff = 11
set throttle_boost = 10
set d_pitch = 58
set f_pitch = 160
set d_roll = 55
set f_roll = 160
set f_yaw = 160
set horizon_level_strength = 30
set d_min_roll = 25
set d_min_pitch = 28
set d_min_boost_gain = 45
set d_min_advance = 0
profile 1

set vbat_pid_gain = ON
set iterm_relax = RPY
set iterm_relax_type = GYRO
set throttle_boost = 10
set f_pitch = 160
set f_roll = 160
set f_yaw = 160
set horizon_level_strength = 30
profile 2

set vbat_pid_gain = ON
set iterm_relax = RPY
set iterm_relax_type = GYRO
set throttle_boost = 10
profile 0
rateprofile 0

set roll_rc_rate = 93
set pitch_rc_rate = 93
set yaw_rc_rate = 93
set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82
set tpa_rate = 38
set tpa_breakpoint = 1650
rateprofile 1

set roll_rc_rate = 93
set pitch_rc_rate = 93
set yaw_rc_rate = 93
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82
set tpa_rate = 38
set tpa_breakpoint = 1650
rateprofile 2

set roll_rc_rate = 92
set pitch_rc_rate = 92
set yaw_rc_rate = 92
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82
set tpa_rate = 80
set tpa_breakpoint = 1800

**Setup / Versions**
 - TBS Colibri Race
 - TBS Bulletproof ESCs with TBS Crossfire

**Additional context**
Was working fine on BF3.5.4 previously

swilkens, 2019-04-22:
Strange, at `FEATURE_CUT_LEVEL 1` the `USE_SERIAL_4WAY_BLHELI_BOOTLOADER` remains enabled. Have you also tried to connect after a clean flash without applying your private settings?

djmartin, 2019-04-22:
Thanks for the reply... 
Yes, tried it with a full reset & clean flash. 
I might try doing a custom performance-build when I get some time through the week to see if I can play with a few other features also (miss my LEDs but can get rid of a few other features) - see if this then behaves or not.
Please let me know if you'd like me to try testing anything else

swilkens, 2019-04-22:
Which ESCs are you using? 3.5.4 was 18 december 2018. The only significant change to the SERIAL_4WAY code was that support for SimonK bootloaders was dropped.

To be clear: Can you connect to the FC using BLHeli suite at all? Is it just the ESC readout that isn't working or can you not connect to the FC using BLHelisuite at all?

djmartin, 2019-04-22:
Hi, this is one of my older quads, but not as old as my good old SimonK ESCs :)  this one is a TBS Powercube V1 ESC stack.  
It doesn't connect at all.
I'll try downgrade when I get some time later in the week - run out of "play time" this long weekend :)

BentyFPV, 2019-04-28:
Hey guys, I'm getting the same issue with the Betaflight F3. Clean flash. Downgrading to 3.5.7 solves the problem. Go back to 4.0 after that and BLHeli32 passthrough no longer works.

etracer65, 2019-04-28:
Try temporarily setting your ESC protocol in Betaflight to MULTISHOT or ONESHOT125 and try connecting. I've seen some intermittent problems using DSHOT (particularly DSHOT1200) when using passthrough mode.

etracer65, 2019-04-28:
Also, can you please confirm which interface you're using in BLHeli suite?

[image]

closed 2019-06-05 by stale[bot]