betaflight/betaflight #8069: 4.0 UART 4 and Turtle mode gone CLRacing f4

OhMyQuadFPV, 2019-04-22:
after flashing betaflight 4.0 my UART 4 was gone, i then set up my modes and there is no 
 "Turtle" mode. I will try to get a diff readout on here when i get a chance. I will also reflash and see if the issue is still there.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

Docteh, 2019-04-22:
Yeah I always try and set up turtle mode before setting the motor protocol to DShot.

OhMyQuadFPV, 2019-04-22:
Reflashing and reinstalling betaflight fixed these issues!

closed 2019-04-22 by OhMyQuadFPV