betaflight/betaflight #8070: synergyf4 target gyro/accel align possible bug

mavericm1, 2019-04-22:
**Describe the bug**
synergyf4 target gyro/accel align is set to cwflip by default.  A test hover seemed to TAZ with this default gyro/accel align on angle mode.   When i set 180 degree flip on roll and set cwflip back to normal everything operated as expected.

**To Reproduce**
Default settings try to hover with angle mode

**Expected behavior**
I expected a normal angle mode hover however it immediately tried to roll right to correct attitude for some reason

**Flight controller configuration**
```
#set gyro_1_sensor_align = CW0FLIP
set gyro_1_sensor_align = DEFAULT
it comes with cw0flip default
and i switched to
#set align_board_pitch = 0
set align_board_pitch = 180
```

**Setup / Versions**
synergyf4 fc 

**Additional context**
n/a

etracer65, 2019-04-22:
As requested, please provide a `diff` of your settings. You haven't even told us what version of Betaflight you're using so I'm not sure how you expect anybody to help.

mavericm1, 2019-04-23:
apologies its betaflight 4.0

```batch start

# reset configuration to default settings
defaults nosave

board_name SYNERGYF4
manufacturer_id KLEE
mcu_id 004b00313436510c35363531
signature 

# name
name RawFPV

# resources
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 A02
resource MOTOR 4 B01

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature ESC_SENSOR

# beeper

# beacon

# map
map TAER1234

# serial
serial 5 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1650 2000 0 0
aux 1 1 0 1800 1900 0 0
aux 2 2 0 1925 2000 0 0
aux 3 13 2 1525 2100 0 0
aux 4 35 2 1700 2100 0 0
aux 5 39 1 900 1300 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_type = PT1
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 200
set dyn_lpf_gyro_min_hz = 110
set dyn_lpf_gyro_max_hz = 575
set acc_calibration = -251,119,269
set rssi_channel = 8
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_bidir = ON
set align_board_pitch = 180
set vbat_max_cell_voltage = 450
set vbat_scale = 101
set yaw_motors_reversed = ON
set small_angle = 180
set transient_throttle_limit = 10
set osd_vbat_pos = 44
set osd_rssi_pos = 2094
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2423
set osd_flymode_pos = 2433
set osd_throttle_pos = 2401
set osd_vtx_channel_pos = 2104
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 2456
set osd_craft_name_pos = 2443
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 2451
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2082
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2251
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 88
set osd_flip_arrow_pos = 2286
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_power = 2
set vtx_freq = 5658
set vcd_video_system = NTSC
set gyro_1_sensor_align = DEFAULT

# profile
profile 0

set dyn_lpf_dterm_min_hz = 100
set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set iterm_relax_cutoff = 35
set p_pitch = 30
set i_pitch = 90
set d_pitch = 27
set f_pitch = 90
set p_roll = 28
set i_roll = 84
set d_roll = 25
set f_roll = 84
set p_yaw = 30
set i_yaw = 90
set f_yaw = 30
set d_min_roll = 16
set d_min_pitch = 18
set d_min_advance = 0
set auto_profile_cell_count = 4

# profile
profile 1

set dyn_lpf_dterm_min_hz = 100
set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set iterm_relax_cutoff = 35
set p_pitch = 30
set i_pitch = 90
set d_pitch = 27
set f_pitch = 90
set p_roll = 28
set i_roll = 84
set d_roll = 25
set f_roll = 84
set p_yaw = 30
set i_yaw = 90
set f_yaw = 30
set d_min_roll = 16
set d_min_pitch = 18
set d_min_advance = 0
set motor_output_limit = 62
set auto_profile_cell_count = 6

# profile
profile 2

set dyn_lpf_dterm_min_hz = 100
set auto_profile_cell_count = -1

# restore original profile selection
profile 1

# rateprofile
rateprofile 0

set roll_rc_rate = 103
set pitch_rc_rate = 103
set yaw_rc_rate = 103
set roll_srate = 66
set pitch_srate = 66
set yaw_srate = 66
set tpa_rate = 75
set tpa_breakpoint = 1400

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save```

etracer65, 2019-04-23:
So what exactly is the problem? The default for this board is `CW0FLIP`. If you have the board mounted in the frame installed the correct side up and the arrow facing forward then no changes should be needed. If you've rotated the board, then just set the board alignment. If you've mounted it upside down, then you'll need to experiment with different sensor alignments (the non "FLIP" versions).

etracer65, 2019-04-23:
And why do this?
```
set align_board_pitch = 180
```
That's effectively going to flip the board so it would make sense that you can't use the correct default sensor alignment of `CW0FLIP`. Assuming you have the board mounted in it's normal orientation (correct side up and arrow facing forward) you should just reset to defaults and stop changing the sensor and board alignments.

mavericm1, 2019-04-23:
the board is mounted in normal orientation the motors are setup with props out and spinning the correct direction and the props were on correctly.  when arming on angle mode it immediately rolled right on the default orientation settings like they were incorrect and flipped out until anti taz kicked in.   after going through all of my settings to be sure that everything was correct changed from cw0flip to 180 degree on pitch and it was fine.   Yes i agree theoretically it should have been working the same but for some reason it wasn't.  I can try the putting the default settings back and try a hover with angle mode on but i highly suspect it would do this again.  I was told to open a issue here to report the problem

mavericm1, 2019-04-23:
[wt-02-bbl.zip](https://github.com/betaflight/betaflight/files/3108091/wt-02-bbl.zip)

etracer65, 2019-04-23:
Just set the sensor and board alignments back to defaults. You can check by looking at the motion in the Setup tab. Just make sure the graphic mirror the physical movements. After that's correct, put the quad on a level surface and calibrate the accelerometer.

mavericm1, 2019-04-24:
the default cwflip seems to be fine with calibration i'll chock this up to my error either not doing a accel calibration or possibly something else i messed up but didn't catch at the time

closed 2019-04-24 by mavericm1