betaflight/betaflight #8071: timerGetByTag is returning the first timerHardware_s entry for non-existent timer pin

jflyper, 2019-04-22:
With `USE_TIMER_MGMT` feature, `timerIndexByTag` is returning `0` for non-existent (non-`IO_TAG_NON`) timer pin, which causes `timerGetByTagAndIndex`to return a pointer to a bogus `timerHardware_s` entry.

Originally reported by: #8066.

jflyper, 2019-04-23:
What is the second exit condition  `timerIndex == 0` at line 67 for?
This is where timerIndex (timer config option) `0` (should mean invalid) is hooked and returning the first entry for a given pin.
https://github.com/betaflight/betaflight/blob/9ad2cc84452b8dcfa385bc5d8ea3cf9ab86ad046/src/main/drivers/timer_common.c#L57-L78

mikeller, 2019-04-25:
@jflyper: I am not 100% sure why it is in there - it was in there when I started working on the code, and I left it in there assuming it was used for something (to get the first option for any given `ioTag`?).

#8093 should fix this.

jflyper, 2019-04-25:
@blckmn Thoughts?

mikeller, 2019-04-28:
@jflyper: I think this was fixed with #8093.

jflyper, 2019-04-28:
Hmm. I was thinking of doing it within `timerGetByTagAndIndex`, but yes, it would do it.

closed 2019-04-28 by jflyper