betaflight/betaflight #8073: motor_output_limit doesn't get properly set when profile switches on auto_profile_cell_count

mavericm1, 2019-04-23:
**Describe the bug**
When flying i noticed motor_output_limit isn't being properly set even though profiles are correct.

iCr4sh [6:41 PM]
It looks like motor limit is set in mixer.c:mixerInit() and initEscEndpoints(), I don't think it's getting called when the profile switches.

**To Reproduce**
profile 0 
set motor_output_limit =100
set auto_profile_cell_count = 4
profile 1
set motor_output_limit = 55
set auto_profile_cell_count = 6
profile 2
set auto_profile_cell_count = -1

plug in 6 cell let quad boot then reboot the fc via osd unplug battery plug in 4 cell take off and fly and you will only get 55% motor output

**Expected behavior**
motor_output_limit to be set in the mixer when the profile changes.

**Flight controller configuration**
```# diff all

# version
# Betaflight / SYNERGYF4 (SYN4) 4.0.0 Apr 11 2019 / 07:46:41 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

# reset configuration to default settings
defaults nosave

board_name SYNERGYF4
manufacturer_id KLEE
mcu_id 004b00313436510c35363531
signature 

# name
name RawFPV

# resources
resource MOTOR 1 A03
resource MOTOR 2 B00
resource MOTOR 3 A02
resource MOTOR 4 B01

# timer

# dma

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL
feature ESC_SENSOR

# beeper

# beacon

# map
map TAER1234

# serial
serial 5 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1650 2000 0 0
aux 1 1 0 1800 1900 0 0
aux 2 2 0 1925 2000 0 0
aux 3 13 2 1525 2100 0 0
aux 4 35 2 1700 2100 0 0
aux 5 39 1 900 1300 0 0

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set gyro_sync_denom = 2
set gyro_lowpass_type = PT1
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 200
set dyn_lpf_gyro_min_hz = 110
set dyn_lpf_gyro_max_hz = 575
set acc_calibration = -251,119,269
set rssi_channel = 8
set serialrx_provider = CRSF
set blackbox_p_ratio = 64
set dshot_bidir = ON
set align_board_pitch = 180
set vbat_max_cell_voltage = 450
set vbat_scale = 101
set yaw_motors_reversed = ON
set small_angle = 180
set transient_throttle_limit = 10
set osd_vbat_pos = 44
set osd_rssi_pos = 2094
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2423
set osd_flymode_pos = 2433
set osd_throttle_pos = 2401
set osd_vtx_channel_pos = 2104
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 2456
set osd_craft_name_pos = 2443
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 2451
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2082
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2251
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 88
set osd_flip_arrow_pos = 2286
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set debug_mode = GYRO_SCALED
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_power = 2
set vtx_freq = 5658
set vcd_video_system = NTSC
set gyro_1_sensor_align = DEFAULT

# profile
profile 0

set dyn_lpf_dterm_min_hz = 100
set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set iterm_relax_cutoff = 35
set p_pitch = 30
set i_pitch = 90
set d_pitch = 27
set f_pitch = 90
set p_roll = 28
set i_roll = 84
set d_roll = 25
set f_roll = 84
set p_yaw = 30
set i_yaw = 90
set f_yaw = 30
set d_min_roll = 16
set d_min_pitch = 18
set d_min_advance = 0
set auto_profile_cell_count = 4

# profile
profile 1

set dyn_lpf_dterm_min_hz = 100
set dterm_lowpass_type = PT1
set dterm_lowpass2_hz = 0
set iterm_relax_cutoff = 35
set p_pitch = 30
set i_pitch = 90
set d_pitch = 27
set f_pitch = 90
set p_roll = 28
set i_roll = 84
set d_roll = 25
set f_roll = 84
set p_yaw = 30
set i_yaw = 90
set f_yaw = 30
set d_min_roll = 16
set d_min_pitch = 18
set d_min_advance = 0
set motor_output_limit = 62
set auto_profile_cell_count = 6

# profile
profile 2

set dyn_lpf_dterm_min_hz = 100
set auto_profile_cell_count = -1

# restore original profile selection
profile 1

# rateprofile
rateprofile 0

set roll_rc_rate = 103
set pitch_rc_rate = 103
set yaw_rc_rate = 103
set roll_srate = 66
set pitch_srate = 66
set yaw_srate = 66
set tpa_rate = 75
set tpa_breakpoint = 1400

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save```

**Setup / Versions**
 motors 2207.5 2450kv  
synergyf4 
aikon 35amp 30x30

rest is n/a

**Additional context**
ICr4sh wrote a quick change for me to test to verify the bug will compile tonight and flash and test some more

iCr4sh [7:07 PM]
Here is a quick fix, if you want to try https://github.com/kmitchel/betaflight/tree/fix_motor_limit_auto_profile

mavericm1, 2019-04-23:
This patch didn't seem to fully resolve the issue.   

i believe this is where the issue is its only looking at the current profile but it takes awhile for the voltage to be detected and switch profiles and by that time the mixer has already init.  In example you ran a 6s battery then you unplug plug in a 4s battery fc boots mixer init i believe is happening before the profile has switched.   then once the profile switches to the 4s profile which is set to 100 motor output it doesn't set the new motor output because the new profile is 100.   There needs to be an elseif that always sets a motor output to 100

https://github.com/betaflight/betaflight/blob/d00a8e7a4c91ed14fd5f5af24c0dcfdca807c2ee/src/main/flight/mixer.c

```
void initEscEndpoints(void)
{
    float motorOutputLimit = 1.0f;
    if (currentPidProfile->motor_output_limit < 100) {
        motorOutputLimit = currentPidProfile->motor_output_limit / 100.0f;
    }
```

mavericm1, 2019-04-25:
adding an else here didn't solve anything either as i should have expected since its being set before the if statement.  As iCr4sh has pointed out the mixer needs to be reinit on auto_profile change but i suspect this would be an issue to make sure it doesn't happen mid flight with a normal profile change on a switch.  rebooting the FC through osd for a second time after the auto profile cell count does properly get the mixer updated with the correct motor_output setting.   Not sure where to go from here i suspect that reinit mixer will need to be added to profile change only if the quad is unarmed.  This is beyond my coding ability.

etracer65, 2019-04-30:
Fixed by #8108

closed 2019-04-30 by etracer65