betaflight/betaflight #8074: Betaflight Motolab Tempest F4 and BF4.0

dcburkhart, 2019-04-23:
**Describe the bug**
Flashed BF4.0 MLTEMPF4 to Motolab Tempest F4 flight controller, configured settings. BLHeli_32 ESC 1 fails to initialize. When checked in BLHeli_32 configurator, ESC 1 not detected. Motors 2, 3 and 4 all spin when armed via Motors tab of BF config and by Radio. I configured from scratch.

**To Reproduce**
Attempted to reflash BF4.0 with same result. Flashed back to BF3.5.1 Target: MLTEMPF4. Configured settings. All 4 ESCs initialize. All 4 motors spin via BF config tab. I am not using nor have I set up the RPM filter feature.

**Expected behavior**
I expected all 4 motors to initialize and spin when commanded to do so.

**Flight controller configuration**
"""# diff

# version
# Betaflight / MLTEMPF4 (MLTE) 4.0.0 Apr 11 2019 / 07:27:20 (9ad2cc844) MSP API: 1.41

# start the command batch
batch start

board_name MLTEMPF4
manufacturer_id 

# name

# resources

# timer

# dma

# mixer

# servo

# servo mix

# feature

# beeper
beeper -ON_USB

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 0 16384 115200 57600 0 115200
serial 1 64 115200 57600 0 115200
serial 3 0 115200 57600 0 115200
serial 4 0 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# display_name

# master
set acc_hardware = NONE
set serialrx_provider = SRXL
set dshot_idle_value = 450
set motor_pwm_protocol = DSHOT1200
set bat_capacity = 1200
set vbat_max_cell_voltage = 440
set vbat_min_cell_voltage = 300
set vbat_warning_cell_voltage = 320
set vbat_scale = 109
set ibata_scale = 155
set yaw_motors_reversed = ON
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1

# profile
profile 0

set vbat_pid_gain = ON

# rateprofile
rateprofile 0

set roll_rc_rate = 72
set pitch_rc_rate = 72
set yaw_rc_rate = 72
set roll_srate = 86
set pitch_srate = 86
set yaw_srate = 86

# end the command batch
batch end

# """

**Setup / Versions**
 - Flight controller; Motolab Tempest F4 v1.0 (purchased from Excel Drones in Garden City)
 - Other components RX: SPM4649T, VTX: Matek VTX-HV, ESCs: DYS Aria BLHeli_32 3-6s 35A(x4), Cobra 2207 2450kv, Camera: Foxeer Predator Mini. Peripheral: Runcam camera control board.
 - how the different components are wired up: Each ESC is wired directly to its factory-assigned output (without resource remapping). VTX is wired to UART 6 for Tramp protocol. Rx is wired to UART t2 for telemetry. Camera is piped through the OSD circuit to VTX. Runcam board is wired from camera OSD control to UART 1. 

**Additional context**
Add any other context about the problem here.

dcburkhart, 2019-04-23:
Yes, I did choose Dshot1200 in the ESC protocol setting.

dcburkhart, 2019-05-03:
Welp,
After attempting to reflash 4.0.1 with varying circumstances, changing the USB cable seemed to fix it. I'm dumb.

dcburkhart, 2019-05-03:
Okay, Not as dumb as we initially thought. After putting the quad back together and reinstalling the SD card for BB logging, the issue resurfaced. I disconnected battery and removed the SD card. Motors arm when connected to battery. It appears that the SD card is preventing m1 from initializing or communicating with the FC, not sure why... or why this is not the case with earlier versions of BF.

dcburkhart, 2019-05-03:
Found that SDcard was stuck in filesystem startup loop. Researched this issue and found that on other FCs some have had similar symptoms (sd card loop, failure of m1 to initialize, works in some BF versions and not others..) Set sdcard_dma = off in cli. This seems to have resolved the issue. Leaving issue open, pending maiden flight and BB log retrieval.

mikeller, 2019-05-03:
@dcburkhart: See comment at the top of https://github.com/betaflight/betaflight/releases/tag/4.0.1

dcburkhart, 2019-05-03:
What am I looking for here?

mikeller, 2019-05-04:
@dcburkhart: Heh, sorry, my bad. I confused your issue with another one that was for an F3 board. Yours is not affected by the F3 issue.

SDCard DMA is known to cause resource conflicts in some setups, which is why it is off by default for all targets but the MOTOLABF4 - for some reason the designer of this target (@MotoLab) decided it was safe to enable it by default.

@MotoLab: What do you think, should we disable SDCard DMA for this target by default, to avoid this type of issue?

dcburkhart, 2019-05-04:
Honestly, it hasn't been an issue until 4.0 and beyond, I'd look for what
changed between 3.5 and now.

MotoLab, 2019-05-04:
DMA worked OK for me on earlier versions, but sdcard_dma on does cause SD card initialization failure on MLTEMPF4 and MLTYPHF4 (MOTOLABF4) targets on 3.5.7 and probably other versions too. If it's off by default on everything else, it should be off on these boards also.

dcburkhart, 2019-05-04:
Yay I'm a contributioner now!

mikeller, 2019-05-04:
@MotoLab: Do you want to open a pull request for this change?

MotoLab, 2019-05-04:
Yes, I will.

mikeller, 2019-06-04:
@MotoLab: Any update on this?

closed 2022-05-10 by haslinghuis