betaflight/betaflight #8075: Betaflight 4.0 OSD not showing VTX Band & Channel

SkyRoxFPV, 2019-04-23:
In BF Config CLI tab

set vtx_band = 5
set vtx_channel = 5
set vtx_power = 1
save

What appear on OSD only **Blank : Blank : 1**, it should be **R : 5 : 1**.

The same thing I wrote on BF 3.5.7 CLI tab, the OSD shows **R : 5 : 1**.

I am using CL Racing F4S FC. All my quad using CL Racing F4S shows the same issue on BF 4.0.

Docteh, 2019-04-24:
Are you sure that your Smart Audio or Tramp is working?

SkyRoxFPV, 2019-04-24:
I just use simple wiring v+, v- & v signal wires. Going back with full chip erase and reset to BF3.5.7 its show the **R:5:1** on OSD, but not showing it with BF4.0.

Docteh, 2019-04-24:
Ah, I've been messing with my uarts today, and it appears as if the OSD element is set via asking the VTX. I can only comment on BF 4.0.0 and master though.

BatesKG, 2019-04-24:
On the Matek F722 build I can confirm the VTX Band/channel is working fine in BF4.0.0.   I'm using an HV Tramp.  
Just to confirm, at a minimum you would need + - power wires, video, and telemetry wires used.   Also in BF4, there is now the option of 3 OSD setups.   Make sure you have the VTX Channel/Power element enabled for the OSD profile you are viewing.  Also note, the power and channel shown is what is stored on the FC and set in the VTX on power up by a TX UART port   IE: if I use my Tramp Wand to change my power/channel after power up, the OSD does not reflect the change and  the FC will revert the VTX to the saved values on the next power cycle.

SkyRoxFPV, 2019-04-25:
Thank you everybody. I am so noob. Just soldered VTX's smartaudio to FC Tx4 solder pad. Set Uart 3 -> TBS Smartaudio. Now its working. OSD show the **R:5:1**. Thanks again guys. Appreciate it very much.

closed 2019-04-25 by SkyRoxFPV