betaflight/betaflight #8077: Free mag alignment in cli

brainbubblersbest, 2019-04-23:
Hello, is it possible to Provider a Free mag alignment in cli? Have a Crocodile lc7 with a 40degree back tilted GPS module Holder. 
On inav Firmware is the exact solution i need
https://github.com/iNavFlight/inav/issues/1029
I will stay on betaflight. But magnetic direction on telemetry data, would be a nice thing. Especially in combination with altitude and distance data If Video cut off. Also a working compass on osd for a long range quad would be fine.  Mag alignment limited to 90 degree and tilt is totally unflexibel if it comes to exotic mag/GPS mounts.

McGiverGim, 2019-04-23:
It is in progress: https://github.com/betaflight/betaflight/pull/7845

brainbubblersbest, 2019-04-23:
Oah, nice 😁 never mind such searchwords. Thank you a lot. Think you can close and erase.

hydra, 2019-04-28:
#7845 is for GYROs, this was for MAG.  re-opened.

closed 2019-06-04 by stale[bot]

ItsMeAubey, 2020-09-19:
The rear tilted magnetometer+gps combo is becoming extremely popular. I think this is still relevant.

brainbubblersbest, 2020-09-19:
It's already implemented.
But instead of align_mag_roll like in inav
The cli Syntax is mag_align_roll.
And in configurator you need to select mag alignment custom.

ItsMeAubey, 2020-09-19:
Oh, perfect! Sorry for the bump.